Barsik-Barbosik/Zoom-Firmware-Editor #2: Zoom G3 limitations?

SrMouraSilva, 2019-05-10:
Hello, 
In the readme it is commented that Zoom G3 is not supported. Is there any limitation or is it a matter of prioritization?

Could you also describe a little bit about the software development process?

Congratulations on the project!

Barsik-Barbosik, 2019-05-13:
Hello,
Zoom G3 is not supported because it has different firmware structure and I don't know how to parse it. I tried to open it and find effect files, but did not found the usual file table there. And it seems to me, that G3 uses smaller pieces of data (smaller than ZDL, which holds all required effect data in one place) and those pieces are more scattered in the firmware updater.

SrMouraSilva, 2019-05-14:
Very cool your research work. What I could suggest is to scan the firmware between G3 version 1 and 2 to look for where new plugins have come to rest on the device. Also compare the B3 zoom firmware with the G3v1. I imagine practically that only patches and plugins change from one version to another.

Barsik-Barbosik, 2019-05-14:
It is a good idea to compare different versions of G3 firmware, but I don't know where to get previous versions. Only the last version (2.10) is available for download.
B3 and G3 firmware have many differences: quick binary comparison with hex editor did not gave positive results. Manual comparison/analysis is required. Later I'm going to choose any effect, that present in both pedals (and maybe exists in ZDL-format too) and search for the common parts of that effect in both firmwares...

SrMouraSilva, 2019-05-15:
From web archive

Zoom G1.12
https://web.archive.org/web/20120809150303/http://zoom.co.jp/download/G3_v1.12_Win_E.zip
Zoom G3 v2.0
https://web.archive.org/web/20130312171152/http://www.zoom.co.jp/download/G3_v2.00_Win_E.zip

Barsik-Barbosik, 2019-05-15:
Thanks for the download links!

therootlord, 2019-05-23:
Ill also do a bin diff

therootlord, 2019-05-25:
> Hello,
> Zoom G3 is not supported because it has different firmware structure and I don't know how to parse it. I tried to open it and find effect files, but did not found the usual file table there. [...]

Hey bro, mine is a G5, i have inspected its firmware, and i found some common patterns probably regarding to the effects data(" And it seems to me, that G3 uses smaller pieces of data"; yep thats true, but i also compared those pieces with some zdl patches and the "core code" seems very similar, they stripped hardly the symbols, debug info etc.. from the firmwares of the old models, so its difficult to associate things)

therootlord, 2019-05-27:
Finding interesting info... It seems they use this http://www.ti.com/tools-software/ccs.html for both old G3 and post devices (ms and Gn series), worth info.. : )

zeroeth, 2020-01-10:
@therootlord I know the g1on/ms series are TMS320c6xxx based https://www.youtube.com/watch?v=iLbD7M8c7Vo I suspect the chip in the G3 is similar if the chip ZFX sticker is removable.