Barsik-Barbosik/Zoom-Firmware-Editor #3: Bass vs. Guitar pedals

mungewell, 2019-06-11:
In your cover page you mention:

> If you are going to inject any GUITAR amplifier effect into any BASS pedal firmware, then you should inject the file "CMN_DRV.ZDL" too!

Can you please explain what you mean?

Is there any physical difference between (say) a G1 Four and a B1 Four (colour excluded), and if I was considering (ab)using one for a Synth set-up would either be preferable to purchase? Just looking at the official docs, the only delta is the effect lists... but you see to have solved that issue.

Thanks,
Simon

Barsik-Barbosik, 2019-06-12:
That sentence regarding "CMN_DRV.ZDL" is relevant for B1On/B1Xon pedals. I should update the readme. That CMN_DRV file is present in guitar firmware (and missing in bass firmware) and it contains a common part of all guitar amplifiers.

I just looked into the contents of "ZOOM G1X FOUR System v1.10 Updater.exe" and there is no such or similar file there. And it seems to me that ZD2-format guitar amplifiers are self-sufficient (like other effects except amps or like bass amps in B1on). But currently, I don't have a Four-series pedal and cannot test it myself. There should be no difference between G1 Four and B1 Four, but just in case, take the pedal that has more suitable effects for you. =)

mungewell, 2019-06-12:
Thanks for the reply, once I've saved up enough pennys I might get/test with a B1Four :-)
Simon.

rmpf, 2019-08-24:
Hi. Yesterday I tried to inject some b1on on my ms70cdr. It did work successfully for the bassman but it never works for the Di5. Do you know why?

hardrubbish, 2019-08-25:
Try this updated version...
https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3465154/ZoomFirmwareEditor_0.99.5.zip