Barsik-Barbosik/Zoom-Firmware-Editor #4: B3n synths imported to G3Xn not available

ctirad, 2019-06-22:
I tried to export/inject two synth effects STDSYN.ZD2 and SYNTLK.ZD2 from B3n to G3Xn firmware (both version 2.20). The process went without any issues, but the two added effects are not available in the G3Xn list. Is it a bug or I am doing something wrong?

mungewell, 2019-07-04:
Looking at the comment on the front page about the G1Four F/W containing "GUARDZDL.ZT2", does the G3X have this too...

The contents of "GUARDZDL.ZT2" just looks to be a list of effect names/files. Perhaps this needs to be updated to list the new effects as well?

Simon.
[still saving my pennies]

mungewell, 2019-07-12:
Hi,
There is a lot of discussion on #7 regarding F/W 2.20, you might want to look there ;-)
Simon