Barsik-Barbosik/Zoom-Firmware-Editor #5: Drum Samples - format and triggering

mungewell, 2019-07-06:
The readme has question marks on the drum sample format. File formats can be easily changed with Sox:
http://sox.sourceforge.net/

ie
`
C:\Users\simon\Downloads\B1 FOUR_v1.10_Win_E\effects>"c:\Program Files (x86)\sox-14-4-2\sox.exe" -r 44100 -e signed-integer -b 16 -c 1 Snr1.raw Snr1.wav
`

Does anyone know whether individual samples can be triggered remotely (via the USB connection)?

mungewell, 2020-05-05:
here's a batch file I used to conert the samples from v2.00:
[convert.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/4581601/convert.zip)

Tomlinsky, 2020-05-08:
I very much doubt they are triggerable externally. There's no sign that the patterns are MIDI data  as far as I remember going through the firmware. Just a simple player with an adjustable clock for deltas to base timings on.

zoomosis, 2020-07-21:
In Linux you can convert all the raw samples to wav, and back to raw with the following two scripts.

This will also work in Windows 10 if you have Linux (eg. Ubuntu) installed under WSL.

You'll need to install "sox" first: "sudo apt get install sox" in Ubuntu.

raw2wav.sh:

```sh
#!/bin/sh

for i in *.raw; do
  sox -t raw -r 44100 -b 16 -c 1 -L -e signed-integer $i `basename "$i" .raw`.wav
done
```

wav2raw.sh:

```sh
#!/bin/sh

for i in *.wav; do
  sox $i -r 44100 -b 16 -c 1 -L -e signed-integer `basename "$i" .wav`.raw
done
```