Barsik-Barbosik/Zoom-Firmware-Editor #6: Will "Large Effects" work on a single display pedal?

mungewell, 2019-07-11:
Reading the manual and specs for the G3n, there are 'standard' and 'large' effects, where the later displays across two of the displays in the pedal.
[image]

I assume that a large effect would not function correctly on a pedal with a single screen (such as G1 Four). Can someone confirm, or deny?

It might be helpful to identify which effects are 'large' and warn when placing them on a single display pedal.

mungewell, 2019-07-11:
Looks like G1Four contains a variant effect, specifically for a single display
```
$ find *_Win_E -name '*800*.ZD2' -exec md5sum {} \;
82664cd7b11d72f9565cfeae097d4f15  G1 FOUR_v1.10_Win_E/effects/MS800_1U.ZD2
bf23cc06adae9ca4920ced3d808cfe12  G3n_v2.00_Win_E/effects/MS800.ZD2
bf23cc06adae9ca4920ced3d808cfe12  G3n_v2.20_Win_E/effects/MS800.ZD2
```

mungewell, 2019-07-11:
Just noticed, the filenames listed in 'GUARDZDL.ZT2' are _NOT_ the ones in the actual firmware, for example it contains 'MS800.ZD2' (not 'MS800_1U.ZD2').

Is this Zoom's way of blocking the effects requiring two displays?

Barsik-Barbosik, 2019-07-22:
Thanks for the interesting information regarding "standard" and "large" effects. I didn't knew about that difference. Unfortunately, I don't have a Four-type pedal for experimenting.

mungewell, 2019-07-25:
Well it just so happens that the Birthday Bunny is bringing me one.... :-)
[image]

Will have to wait before opening it, but will update once I start hacking it.

mungewell, 2019-08-19:
I've also noticed that the ZT2 files for the G1N vs G3N show that there are '3S' effect names too.... suspect that these are variants specifically for the G1N (perhaps optimized for processor and/or memory configuration).

Note: ID appears to be '+1' vs the G3N.
```
C:\Users\simon\Documents\ZoomFW>diff --side-by-side "G1 FOUR_v1.10_Win_E\zt2\FLST_SEQ.ZT2.txt" G3n_v2.20_Win_E\zt2\FLST_
SEQ.ZT2.txt
...
                Container:                                                      Container:
                    effect = u'GT_GEQ1U.ZD2' (total 12)       |                     effect = u'GT_GEQ.ZD2' (total 10)
                    version = u'1.00' (total 4)               |                     version = u'1.10' (total 4)
                    installed = 0                             |                     installed = 1
                    id = 81                                   |                     id = 80
                    unknown = 0                                                     unknown = 0
                    group = 2                                                       group = 2
                Container:                                                      Container:
                    effect = u'GTGEQ71U.ZD2' (total 12)       |                     effect = u'GT_GEQ_7.ZD2' (total 12)
                    version = u'1.00' (total 4)               |                     version = u'1.10' (total 4)
                    installed = 1                                                   installed = 1
                    id = 84                                   |                     id = 83
                    unknown = 0                                                     unknown = 0
                    group = 2                                                       group = 2
                Container:                                                      Container:
                    effect = u'STGTGE1U.ZD2' (total 12)       |                     effect = u'ST_GT_GE.ZD2' (total 12)
                    version = u'1.00' (total 4)               |                     version = u'1.10' (total 4)
                    installed = 0                             |                     installed = 1
                    id = 97                                   |                     id = 96
                    unknown = 0                                                     unknown = 0
                    group = 2                                                       group = 2
...
                Container:                                                      Container:
                    effect = u'CRNTRI3S.ZD2' (total 12)       |                     effect = u'CRN_TRI.ZD2' (total 11)
                    version = u'1.00' (total 4)               |                     version = u'1.20' (total 4)
                    installed = 1                                                   installed = 1
                    id = 305                                  |                     id = 304
                    unknown = 0                                                     unknown = 0
                    group = 6                                                       group = 6
...
                Container:                                                      Container:
                    effect = u'LPROLL3S.ZD2' (total 12)       |                     effect = u'LOOPROLL.ZD2' (total 12)
                    version = u'1.00' (total 4)               |                     version = u'1.10' (total 4)
                    installed = 0                             |                     installed = 1
                    id = 65                                   |                     id = 64
                    unknown = 0                                                     unknown = 0
                    group = 7                                                       group = 7
```

[diff.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3513968/diff.txt)

mungewell, 2019-09-08:
Closing this out as this seems to be confirm. Specifically ZOOM have blacklisted some effects on the G1Four/G1Four-X/B1Four/B1Four-X.

closed 2019-09-08 by mungewell