Barsik-Barbosik/Zoom-Firmware-Editor #7: Compatibility with v2.20 firmware for G3n/B3n?

nairanvac, 2019-07-11:
Is the firmware editor software compatible with version 2.20 of the firmware for the G3n and B3n pedals?  None of the effects I've added have shown up after flashing the modified firmware.  I've tried to find version 2.00 to test it with, but it doesn't seem to be available online.

mungewell, 2019-07-11:
Got to love the Internet Archive (go make a dontation ;-)
V2.0
https://web.archive.org/web/20181022055543/https://www.zoom.co.jp/sites/default/files/products/downloads/software/G3n_v2.00_Win_E_0.zip

mungewell, 2019-07-11:
There is reference to "FLST_SEQ.ZT2" in the firmware, but this 'file' is not displayed by the Editor.

On other firmware the "ZT2" file appears to be a list of available effects. Don't know if it is possible to recreate this and upload with the extra effects.
```
$ hexdump -C *.ZT2 | head
00000000  47 54 5f 47 45 51 2e 5a  44 32 00 00 00 47 54 5f  |GT_GEQ.ZD2...GT_|
00000010  47 45 51 5f 37 2e 5a 44  32 00 53 54 5f 47 54 5f  |GEQ_7.ZD2.ST_GT_|
00000020  47 45 2e 5a 44 32 00 45  47 46 49 4c 54 45 52 2e  |GE.ZD2.EGFILTER.|
00000030  5a 44 32 00 54 53 5f 42  4f 4f 53 54 2e 5a 44 32  |ZD2.TS_BOOST.ZD2|
00000040  00 52 45 44 43 52 42 53  54 2e 5a 44 32 00 4d 53  |.REDCRBST.ZD2.MS|
00000050  38 30 30 2e 5a 44 32 00  00 00 00 4d 53 31 39 35  |800.ZD2....MS195|
00000060  39 2e 5a 44 32 00 00 00  4a 54 4d 34 35 4f 53 2e  |9.ZD2...JTM45OS.|
00000070  5a 44 32 00 00 46 44 5f  54 57 49 4e 52 2e 5a 44  |ZD2..FD_TWINR.ZD|
00000080  32 00 46 44 5f 42 4d 41  4e 2e 5a 44 32 00 00 46  |2.FD_BMAN.ZD2..F|
00000090  44 5f 44 4c 58 52 2e 5a  44 32 00 00 46 44 5f 4d  |D_DLXR.ZD2..FD_M|`

nairanvac, 2019-07-11:
> directly typed link works
> https://www.zoom.co.jp/sites/default/files/products/downloads/software/G3n_v2.00_Win_E.zip

Thank you so so much for finding that link!  I'll try it tonight and let you know how it goes.  I find it interesting that v2.20 was released right around the time this firmware editor was released, and there don't seem to be any major changes in this version.  Makes me wonder if it was released specifically to hamper this tool.

mungewell, 2019-07-11:
I have little faith that a large organization could move so fast.... ;-)

On the topic of ""FLST_SEQ.ZT2", it looks like it's being excluded from the listing, but I still think (with no proof/experience) that it might be helpful to recreate it when add new effects.
https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/blob/826654efc5d3743482b35d9acfabc9b514eee2ff/src/main/java/zoomeditor/model/Firmware.java#L26

I presume you have double-checked that the new effects are listed if you re-edit the modified f/w...

nairanvac, 2019-07-11:
Looks like downgrading to 2.00 fixed it. Everything is working fine now. I have all the guitar effects I want in my B3n now.

And yes, I did double check that the effects were appearing in the modified 2.20 firmware by loading it into the editor. Good thought.

mungewell, 2019-07-12:
Since I'm on the verge of buying a G3n very happy that you confirm that it can work (albeit with older f/w).

Not a Java programer so not sure how to re-enable the extraction of the "ZT2" files, but I did use a binary editor on the EXE to rename ZT2 -> ZT3 and I have it :-)

The file does list the effects, but in groups which presumably match the type of effects. Also fairly experienced with parsing/builiding files with Construct. Just need to figure out the structure.

Which effects did you add, perhaps we can try editing the ZT2 and re-insert it.

mungewell, 2019-07-12:
And with some Python... obviously I guessed at some of the characteristics of the file (which came from v2.20).
```
C:\Users\simon\Downloads\G3n_v2.20_Win_E\zt2>"c:\Program Files\Python36\python.exe" zoom_zt2.py -t 2 -w -d FLST_SEQ_MOD.ZT2
Reading FLST_SEQ_MOD.ZT2...
Adding to Group 2:
ListContainer:
    Container:
        name = u'BYPASS.Z' (total 8)
    ListContainer:
        Container:
            group = 1
            effects = ListContainer:
                Container:
                    effect = u'COMP.ZD2' (total 8)
                    version = u'1.20' (total 4)
                    id = 16
                    unknown = 0
                    group = 1
...
            group2 = 1
        Container:
            group = 2
            effects = ListContainer:
                Container:
                    effect = u'AUTOWAH.ZD2' (total 11)
                    version = u'1.20' (total 4)
                    id = 16
                    unknown = 0
                    group = 2
...
                Container:
                    effect = u'FAKEME.ZD2' (total 10)
                    version = u'1.23' (total 4)
                    id = 224
                    unknown = 0
                    group = 2
            group2 = 2
        Container:
            group = 3
```
[zoom_zt2.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3385090/zoom_zt2.zip)

nairanvac, 2019-07-12:
Downgrading to 2.00 worked just fine, and now I'm able to edit the firmwares effectively.  Thank you so much, @mungewell!

I'll leave this issue open for now since it seems we're still unsure about compatibility with v2.20.

nairanvac, 2019-07-12:
> And with some Python... obviously I guessed at some of the characteristics of the file (which came from v2.20).

> [zoom_zt2.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3385090/zoom_zt2.zip)

So, you're saying I should be able to put this file into the B3n v2.20 firmware and be able to use effects from the G3n?

It's worth noting that I started out using v2.20 of both the G3n and B3n firmware.  I have not tried importing any effects between versions.

mungewell, 2019-07-12:
What I am saying(/speculating) is that F/W 2.20 may not work if the "FLST_SEQ.ZT2" doesn't match the extra effects which have been loaded.

Say you add 'AUTOWAH.ZD2' to a F/W that didn't have it, then you might need to add a section to "FLST_SEQ.ZT2" with something like
```
                Container:
                    effect = u'AUTOWAH.ZD2' (total 11)
                    version = u'1.20' (total 4)
                    id = 16
                    unknown = 0
                    group = 2
```

This would mean having access to the "ZT2" for _BOTH_ F/W, firstly to find out what the values are for the effect which is being copied and then secondly to insert them into the file for the target/modified F/W.

mungewell, 2019-07-14:
I updated my script and included a pre-built version (for Win10 64bit)
[zoom_zt2_20190713.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3389386/zoom_zt2_20190713.zip)

You can now add/remove effects to the "FLST_SEQ.ZT2" file at will.
_Disclaimer is that I don't actually have a pedal to test this on.... ordering one for my birthday :-)_
```
$ python3 zoom_zt2.py --help
Usage: zoom_zt2.py [options] FILENAME

Options:
  -h, --help            show this help message and exit
  -d, --dump            dump configuration/sequence to text
  -A ADD, --add=ADD     add an effect to group ADD
  -n NAME, --name=NAME  effect name (use with --add)
  -v VER, --ver=VER     effect version (use with --add)
  -D DELETE, --delete=DELETE
                        delete last effect in group DEL
  -w, --write           write config back to same file
```

For example to add the TapeEcho effect you would need
```
$ python3 zoom_zt2.py -A 8 -n 'TAPEECHO.ZD2' -ver "1.30" -w FLST_SEQ.ZT2
```

The '-D' flag will cause the last effect to be removed from a group, should that be needed. There may be some weirdness with the very last group/last effect as that seems have a strange ID number. Last group is 'PEDAL' for the G3n/B3n.

Barsik-Barbosik, 2019-07-22:
> Not a Java programer so not sure how to re-enable the extraction of the "ZT2" files, but I did use a binary editor on the EXE to rename ZT2 -> ZT3 and I have it :-)
Nice workaround. :-) It is possible to disable excluding of ZT2-files by setting excludeSequenceFiles=false in the [config](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/blob/master/src/main/resources/app.config).
Can anyone confim, that editing of "FLST_SEQ.ZT2" is necessary in the latest G3n/B3n firmwares? Than I will add that logic into java code.

ribastudio, 2019-08-10:
Hi people, i'll try. Can u tell me if the Zoom's G3n and B3n have the same hardware ?

Inside G3n FLST_SEQ.ZT2 (ver 2.20), have the respective ids:

Zoom G3n v2.20

01: DYN
02: FLTR
03: GUITAR OVERDRIVE/DISTORTION
04: GUITAR AMP SIMULATOR
05: CABINET
06: MODULATION
07: SFX
08: DELAY
09: REVERB
0B: PEDAL

I've inserted the bass effects inside the guitar groups.

mungewell, 2019-08-10:
I would be very surprised if the boards aren't exact the same.

Are you using the python script to modify ZT2? The firmware for B3n is also available, so you can dump it's ZT2 file to see were Zoom places each effect.
[FLST_SEQ.ZT3.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3489220/FLST_SEQ.ZT3.txt)

I tried the latest JAR file on #11 but it didn't appear to automatically change ZT2. And I have just over a week before I can open my present... I wonder what it could be ;-)

ribastudio, 2019-08-21:
I've tried, edited the ZT2 but the effects don't appear at all :( I wanna amps and two effects to usae with my bass, and the rest to my guitar in my productions

mungewell, 2019-08-21:
I believe the OP downgraded to V2.00 and had success.

I have also been attempting to get the G1Four running (https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/issues/14#issuecomment-522866949) and have an update Python script there. Still no success though. :-(

Can you confirm what/how you changed FLST_SEQ, and whether you have been using Guitar Lab with your panel (just a hunch that it might complicate things).

ribastudio, 2019-08-21:
I've changed with ur script that i've downloaded here. I extract form B3n firmware, looking a B3N file structure (heads, etc) and realign in a the best firware of both worlds. But, still not functioning
And, yes, I'm using GuitarLab. If have an other program to maintain a programming easy, i'm using that...

hardrubbish, 2019-08-21:
I doubt it'd make a difference, but have you tried using ToneLib instead of GuitarLab?
https://tonelib.net/zoom-quick-start/

mungewell, 2019-08-21:
The reason I mentioned Guitar Lab is the 'weirdness' (probably desired by most people) of maintaining the installed/not-installed state of the effects when uploading F/W (I went from V1.00 to V1.10 on my G1Four, see #14 ).

It's almost as if it's maintaining a list of effects somewhere on the pedal, other than the FLST_SEQ file (embedded in f/w). I spent a while last night seeing if I could snoop the communications between Guitar Lab <-> pedal when installing/removing effects, but failed to find a method.

ElPuni, 2020-06-19:
I have a B3n and did some tests to add patches from both G3n and G5n. I could confirm as you say with firmware versions 2.20 patches didn´t appear in the pedal, but using firmware 2.00 did the trick, at least injecting patches from G3n. I´ve been trying to find on the internet (if it existed) a G5n 2.00 firmware version to see if injection also works, but I cannot find it anywhere, only version 3.00. Can anybody share/point to an address where I can download Zoom G5n firmware 2.00 or older version to try? Thanks in advance.