Barsik-Barbosik/Zoom-Firmware-Editor #8: Java trial runtime errors?

nairanvac, 2019-07-13:
Today the firmware editor stopped working, giving an error about a trial installation of a Java runtime.  I attempted to reinstall, but was unable after receiving the error "This trial installation has expired."

Just wanted to give a heads up.  So glad I got it to work before it broke!

mungewell, 2019-07-14:
I got the same today....
[image]

The work around is to install the java runtime and they download the '.jar' of the project with
```
"c:\Program Files (x86)\Java\jre1.8.0_211\bin\java.exe" -jar ZoomFirmwareEditor.jar
```

Barsik-Barbosik, 2019-07-22:
Thanks for reporting! That unexpected problem came from demo-version of Excelsior Jet compiler, that was used to create a Windows exe-file from java code. I didn't knew that it had a time-limit.
Tomorrow I will create a new executable, using the full version of Excelsior JET 15.3, Standard Edition.

Barsik-Barbosik, 2019-07-23:
[ZoomFirmwareEditor-setup.exe](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/releases/download/v0.98/ZoomFirmwareEditor-setup.exe) has been updated. Now it does not have a time-limit.

closed 2019-07-23 by Barsik-Barbosik