Barsik-Barbosik/Zoom-Firmware-Editor #9: Where is the ZoomFirmwareEditor-setup.exe file?

Gui-GCCarvalho, 2019-07-21:
Hi, I was trying to install the firmware editor and i can't put to work running with java. I went by the ZoomFirmwareEditor-setup.exe file and did not find it.

Someone have this executable file?

Barsik-Barbosik, 2019-07-22:
Hi, you can download it from "[releases](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/releases)" page. Or use a direct [download](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/releases/download/v0.98/ZoomFirmwareEditor-setup.exe) link.
Unfortunately, the current executable file has a time-limit (see https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/issues/8), I am going to fix that tomorrow. ZoomFirmwareEditor.jar does not have that problem.

Gui-GCCarvalho, 2019-07-22:
Hi!

I’m using Mac and can’t use the .java file.

I Will try your sugestion, thx

Enviado do meu iPhone

Barsik-Barbosik, 2019-07-23:
That link could be useful for Mac-users: [How to install Java on Mac](https://stackoverflow.com/questions/24342886/how-to-install-java-8-on-mac).

Gui-GCCarvalho, 2019-07-23:
Do you know why the G1on AK (andreas kisser version) show the message “illegal device” when I try to make the update with modified firmware?

Enviado do meu iPhone

closed 2019-07-26 by Barsik-Barbosik