Barsik-Barbosik/Zoom-Firmware-Editor #10: Why the G1on AK version don`t work??

Gui-GCCarvalho, 2019-07-23:
Hi!

Someone has sucess at the use of firmware editor with G1on AK version?

When i`m going to make the update show a message "illegal device".

Barsik-Barbosik, 2019-07-23:
Do I understand correctly, that you have modified the firmware updater (which previously worked without errors) and after that it does not update the pedal, instead shows an "illegal device" message? Did you updated the special **G1on AK** firmware updater or the usual G1on firmware updater? (I am asking, because I don't see the G1on AK firmware on the Zoom official site).

There can be the same [issue](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/issues/7), like with latest versions of G3n/B3n firmware: FLST_SEQ file update is required.

Gui-GCCarvalho, 2019-07-23:
I don`t found the firmware for G1on AK version. I tried the standard G1on
firmware and doesn`t works.

How can I solve this problem?

Gui-GCCarvalho, 2019-07-23:
I tried to fix with the FLST_SEQ but I can`t put this file at the firmware
editor because the mac version of 1.21 firmware G1on doesn`t is valid

like the print

When I try the Windows version, I can`t put the file ante firmware editor
like the second print

Barsik-Barbosik, 2019-07-24:
Sorry, I don't know how to help you with that trouble :-( One option is just wait until Zoom Corporation releases the firmware updater for the G1on AK-version...