Barsik-Barbosik/Zoom-Firmware-Editor #11: MultiStomps not displaying injected effects.

hardrubbish, 2019-07-26:
What an exciting project! I’ve been hoping someone might do something like this after Zoom added the extra effects to the 50G. Finally, it’s happening!

I’ve tried this on two 70CDR and a 50G. I’ve tried the editor on a windows 10 computer and a Mac 10.13. 
On both 70CDR they didn’t show the injected effect, but would freeze when scrolling up when it got to the Bass Equaliser effect.
The 50G didn’t show anything but didn’t freeze. 
I was trying to inject BassMuff from the 60B.
Anything else I could try?

Barsik-Barbosik, 2019-07-26:
Thanks for the reporting!
I have an idea, what may be wrong: removing of FLST_SEQ.ZDT. That file is not used on G1on/B1on, but it may be required for multistomps. I'm going to fix it as soon as possible.

hardrubbish, 2019-07-26:
Ah great! Hopefully that does the trick.
I’ll give it another shot when you have time to update. 👍

Barsik-Barbosik, 2019-07-28:
_Some technical info related to FLST_SEQ._ 

Effects should be inserted into the correct group inside the FLST_SEQ. There are 11 different ZDL-effect types/groups:
- 01: DYN
- 02: FLTR
- 03: GUITAR OVERDRIVE/DISTORTION
- 04: GUITAR AMP SIMULATOR
- 05: BASS AMP SIMULATOR
- 06: MODULATION
- 07: SFX
- 08: DELAY
- 09: REVERB
- 0B: PEDAL
- 0C: BASS OVERDRIVE/DISTORTION

Effect type can be read from ZDL-file.

[image]

hardrubbish, 2019-07-29:
Ah, I see. So there's a specific order that the effects need to be in... Is this universal? 
For example, the 70CDR has no amps or overdrive/distortion. Is the Zoom still going to expect any of those added with the editor to be in the order as you've shown?

Barsik-Barbosik, 2019-08-01:
I just added the FLST_SEQ.ZDT file update. I hope that problem with multistomps is now solved. Could you please download that special version and try with 70CDR and 50G?
[ZoomFirmwareEditor_ZDT.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3458904/ZoomFirmwareEditor_ZDT.zip)
Please ensure that FLST_SEQ.ZDT exists in the file list before performing the update.

hardrubbish, 2019-08-02:
Ah, terrific! Thanks for that.
I'll have a crack now.

hardrubbish, 2019-08-02:
Ok, I have tried that. It seems to have an identical outcome as the original. 
FLST_SEQ appears at the bottom of the list, and I've injected the new ZDL file near the bottom there...

[image]

hardrubbish, 2019-08-02:
I just noticed that the effects display in a different order on the pedal's display with the official firmware and modified one. The effects are in the new order,  Z-Tron is down and Corona is up, whereas in the original Stereo Bass is down from the THRU page and Tremolo is up.

I also tried installing the modified firmware without any extra effects injected. It doesn't freeze when you cross over the THRU page. 

I hope some of this might help.

Barsik-Barbosik, 2019-08-02:
Thank you very much for the testing and feedback!
I'll continue with that sequence file. I have noticed that there are 2 more possible types of effects (14 and 16: both are bass drive types). And there are 2 versions of BassMuff effect: BASSMUF.ZDL (from MS-60B, type is "0C") and _BASSMUF.ZDL (from B1on, type is "14").

Barsik-Barbosik, 2019-08-02:
I think, that BASSMUF.ZDL effect version requiers the another file "CMN_BASS.ZDL"!! I didn't knew that such file exists in the MS-60B firmware! :-)
[image]

Barsik-Barbosik, 2019-08-02:
[ZoomFirmwareEditor_0.99.2.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3462704/ZoomFirmwareEditor_0.99.2.zip) - version 0.99.2 (same as previous, but supports 13 effect types).

Barsik-Barbosik, 2019-08-02:
Interesting... BASSMUFF.ZDL from MS-60B makes my B1Xon freeze. Alone or with the CMN_BASS.DRV. Finally I am able to reproduce the bug. :-)
I will try to investigate, why this happens. Currently I suggest to import bass effects from B1on/B1xon firmware.

hardrubbish, 2019-08-03:
Thanks for all the effort on this! I feel like we're getting closer.
I've tried 0.99.2, no freezing this time! But it's still not displaying the injected effect. I'm trying this on the CDR using the bassfuzz from the b1on.

Barsik-Barbosik, 2019-08-03:
[ZoomFirmwareEditor_0.99.3.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3463707/ZoomFirmwareEditor_0.99.3.zip) - with "effect type" column.
Now I will make 2 more experimental versions, especially for CDR.

Barsik-Barbosik, 2019-08-03:
Could you try to inject that effect?
[_BAZZMUF.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3463737/_BAZZMUF.zip)
This is the _BASSMUF.ZDL from B1Xon, but with replaced effect type.

Barsik-Barbosik, 2019-08-03:
[ZoomFirmwareEditor_0.99.4_CDR.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3463757/ZoomFirmwareEditor_0.99.4_CDR.zip) - experimental version for CDR, theoretically should work with [_BAZZMUF.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3463737/_BAZZMUF.zip)

hardrubbish, 2019-08-03:
Ah great! I'll give these a go and get back to you.
Fingers crossed!
Thanks.

hardrubbish, 2019-08-03:
Success! 0.99.3 with your _bazzmuff zip worked.
No freezing, and the injected effect appears!

hardrubbish, 2019-08-03:
Ahh, hang on. Your _bassmuff file is appearing differently. When I extract the effects from the various firmwares, the system doesn't recognise the filetype, whereas with yours it does.
[image]
Yours on top, mine below.

hardrubbish, 2019-08-03:
So, just making sure I’m doing this right and not wasting your time here…

To extract the effect/s I load the ZoomFirmwareEditor .jar file.
Open the firmware that I want to extract the effect/s from.
Select effect/s from the list, click Extract, then designate which folder I want to save the effect to.

All those ZDL files are coming up as unrecognised.
[image]

Barsik-Barbosik, 2019-08-03:
> Success! 0.99.3 with your _bazzmuff zip worked.

Nice! Finally! :-)

> Your _bassmuff file is appearing differently. When I extract the effects from the various firmwares, the system doesn't recognise the filetype, whereas with yours it does.

Then all bass effects should be converted before injecting into 70CDR. Just 1 byte should be replaced.
[image]

Tomorrow I will prepare the new version with authomatic conversion.

hardrubbish, 2019-08-03:
Oh right! I didn’t realise that I should have been editing the hex too. 
Although, all the effects that I’ve extracted, even the CDR ones aren’t being recognised by the computer. So they need to have the hex edited too, once extracted?

Barsik-Barbosik, 2019-08-04:
[ZoomFirmwareEditor_0.99.5.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3465154/ZoomFirmwareEditor_0.99.5.zip) - added automatic type replace for multistomp pedals.

> all the effects that I’ve extracted, even the CDR ones aren’t being recognised by the computer

These effect files are usually not recognised by the computer and it's normal. But you can use any hex-editor in order to view their content (if I understood you question correctly).

hardrubbish, 2019-08-04:
Oh dear, that's back to freezing and not displaying the injected effect.
I re-extracted effects using the new editor version.

What I meant was, there was something different about the bazzmuff effect that you'd extracted and how my computer was reading it. The Mac OS was giving it an "exec" icon, whereas everything I've extracted is a blank, unrecognised file icon. 
If the new editor version was doing the automatic conversion but still not working, perhaps there's something about how the effects are being extracted... Puzzling.

hardrubbish, 2019-08-05:
Not sure if this'll be any help, but here's the file as I've extracted it.
[BASSMUFF.ZDL.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3465571/BASSMUFF.ZDL.zip)

Barsik-Barbosik, 2019-08-05:
But your BASSMUFF.ZDL is extracted from MS60B... Please, try effects from Zoom B1on/B1Xon. :-)

hardrubbish, 2019-08-05:
Ah right. Sorry, I forgot about that bit...
So yes, that works!

hardrubbish, 2019-08-05:
So, on the 70CDR it only seems to accept ZDLs from the B1ON. Not G1ON or MS50G.
Haha, sorry... I'm probably doing something wrong here.

Barsik-Barbosik, 2019-08-05:
> So, on the 70CDR it only seems to accept ZDLs from the B1ON. Not G1ON or MS50G.

It should accept guitar effects too. ZDL-files from MS50G and G1on are identical. If you have a problem with any ZDL from  G1ON or MS50G, then please write the file names. That will help a lot!
Did you injected "CMN_DRV.ZDL" as well (that file is required for all guitar amps)?

And once again, thank you very much for helping with testing!

hardrubbish, 2019-08-06:
Ok. Sorry aboyut the delay. I'm back with some more time to tinker. 
I'm on a win10 laptop now. I got the CDR to load 60B effects! 
I included the CMN_BASS.ZDL file which is listed as TYPE "0f" in the "type" column and seems to have done the trick.
No freezing, everything's showing up!
I'll keep mucking around and will let you know if anything freezes up again, but it's looking pretty good now.
Thanks for sorting that out!

hardrubbish, 2019-08-06:
Ha, I spoke too soon. So, I managed to get the _B_FZSML, and _B_SQUEK on. Then added _BASS_TS, BASSDRV and BASSMUFF. Then it started to freeze again. I then took them back off one by one until it stopped freezing. Seems like the BASSDRV is a problem too.

_GKruger amp is fine. MarkB freezes. I still have CMN_BASS injected. B_AtWah freezes it. Seems like 60B ZDL effects that don't start with an underscore will freeze it.

I'll have a much around with the B1ON effects now.

hardrubbish, 2019-08-06:
I've noticed that the order you put things in doesn't seem to make a difference to how they display on the pedal. I'm just dumping them all down the bottom and they're listing in an order, probably the effect order you posted above.

Barsik-Barbosik, 2019-08-06:
> I've noticed that the order you put things in doesn't seem to make a difference to how they display on the pedal. I'm just dumping them all down the bottom and they're listing in an order, probably the effect order you posted above.

Yes, the effects are currently ordered simply by their types: 01, 02, 03, 0C, 0D, 14, 16, 04, 05, 06, 07, 08, 09, 0B. I'm still thinking how to fix the functionality of "Move up" and "Move down" buttons. They should move the effect only inside the appropriate group. And the initial effects order may be read from the FLST_SEQ.ZDT list (but not all files described there: drum samples, cmd_drv etc).

> _GKruger amp is fine. MarkB freezes. I still have CMN_BASS injected. B_AtWah freezes it. Seems like 60B ZDL effects that don't start with an underscore will freeze it.

Today I'm going to experiment with MS60B bass effects on my B1Xon pedal: I'll try to replace their 
 type (same 1 byte), but vice versa. Maybe they will start to work. :-)

hardrubbish, 2019-08-06:
You could you have a column for the effect types, and then set the column to be adjustable  alphabetically and numerically via the column title tab. Oh, but that would only change the order according to the firmware file list, not how the pedal is reading and listing the effects internally.
The Move Up/Down buttons didn't seem to be necessary the last time I was compiling. The injected effects would sit at the bottom of the list, but the pedal would still display them in that predetermined way. 
It would be handy to have some control over the ordering of the "Effect Type" column, so you can see effect types all collected together rather than somewhat scattered as they are now.

Good luck with the MS60B! Let me know how you go with it.

mungewell, 2019-08-10:
Glad to see progress, but can you confirm whether latest version (as above https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/issues/11#issuecomment-518024269) should automatically change the ZT2 file - or do we need to manual do it and re-upload to F/W?

For me it seems to stay the same
```
C:\Users\simon\Documents\ZoomFW\G1 FOUR_v1.10_Win_E>md5sum mod_FLST_SEQ.ZT2
90b650759ead457dfe7957e0d2fb5bd0 *mod_FLST_SEQ.ZT2

C:\Users\simon\Documents\ZoomFW\G1 FOUR_v1.10_Win_E>md5sum zt2\FLST_SEQ.ZT2
\90b650759ead457dfe7957e0d2fb5bd0 *zt2\\FLST_SEQ.ZT2
```

insektgo, 2019-08-10:
I downloaded the ZoomFirmwareEditor_0.99.5.zip version (Latest I could find here) and used it to extract the DISTPLUS.ZDL, FUZZSML.ZDL, and GRTMUFF.ZDL files from the "ZOOM MS-50G System v3.10 Updater.exe" file... I then opened the "ZOOM MS-70CDR v2.00 Updater.exe" and injected the new .ZDL files. I loaded the "MODIFIED ZOOM MS-70CDR v2.00 Updater.exe" into the MS-70 CDR and the injected .ZDL files were not present. Not sure what else I might try to get this to work. Love the idea on this project... Love MS-70 CDR but I'd like to get some Fuzz and Distortion out of it as well. Thanks for all your hard work.

insektgod, 2019-08-14:
Injecting the CMN_DRV.ZDL file along with the other files extracted from the MS-50 G firmware worked. I finally have Fuzz and Distortion on the MS-70 CDR. I couldn't be happier.

Barsik-Barbosik, 2019-08-16:
Thanks for the confirmation, that MS-50 fuzz and distortion work on CDR-70.

> Glad to see progress, but can you confirm whether latest version (as above #11 (comment)) should automatically change the ZT2 file - or do we need to manual do it and re-upload to F/W?

ZT2 creation is not ready yet. I will continue to implement that part during the next week (currently don't have time).

soymonitus, 2020-02-11:
Hi, I would like to flash in the mS50g the effects from the ms70cdr that are missing like all those prefixed with "Ba". How should I do it?

hardrubbish, 2020-02-11:
I had this same issue. There’s something odd about the MS bass effects. It was suggested to source them from the B1ON instead. That allowed me to inject bass effects onto my 70CDR.

kielichwina, 2021-03-19:
> Thanks for the confirmation, that MS-50 fuzz and distortion work on CDR-70.
>
>
> ZT2 creation is not ready yet. I will continue to implement that part during the next week (currently don't have time).

any news about ZT2 conversion?

nomadbyte, 2023-06-27:
In case anyone still cares to have the "new" (as non-B1on style)  MS-60B Bass AMP and DRV effects, it's possible to make them function on MS-50G/MS-70CDR; effects like `SVT.ZDL` or `B_OD.ZDL`, those without the leading underscore.

These "new" effects make use of common functionality, which was split off into `CMN_BASS.ZDL` module. This saved some flash space, as such effects are slimmer. However, this required the firmware to load up this common module (likely on startup as it seems to be a plug-in design). While MS-60B firmware loads up both `CMN_BASS.ZDL` and `CMN_DRV.ZDL`, the MS-50G/MS-70CDR firmware load only the `CMN_DRV.ZDL`. This was all of the "new"  Bass AMP and DRV effects simply lack the needed code loaded, even though the `CMN_BASS.ZDL` file is present on the flash.

So to get around this, it's possible to simply "inject" the `CMN_BASS.ZDL` under the name `CMN_DRV.ZDL`. This definitely works for the Bass DRV effects. It also will properly load the Bass AMP effects (they will appear for selection and edit) ... but without CABs, that is the CAB parameter is not set, can only be properly set to OFF.

Obviously, this approach excludes all of the MS-50G Guitar AMP and DRV effects, because they need the "real" `CMN_DRV.ZDL`.

The deal about the CAB is that the Bass CAB is tagged in ZDL as __BCAB__, that's what the MS-60B firmware is expecting. Yet the `CMN_DRV.ZDL` is used with Guitar AMP ZDL which tag the CABs as __CABI__.  So technically it's possible to change `BCAB` to `CABI` in the "new" Bass effect's ZDL, but that further "chimerizes" this while set up.

Not sure how viable it is to alter this directly in the MS-50G/MS-70CDR "custom" firmware without danger of bricking the pedal.

Another alternative is to consider merging both `CMN_DRV.ZDL` and `CMN_BASS.ZDL` -- both are dynamic libraries, so theoretically it may be possible to copy the needed entries, though it's quite prone to error.

nomadbyte, 2023-06-27:
As for the "underscored" MS-60B Bass effects (like `_SMR.ZDL`), those can function on MS-50G/MS-70CDR even without `CMN_DRV.ZDL` present. However, such AMP effects won't have any CAB, basically it's OFF, even though the screen still shows some specific cab names.

Again, the issue here also seems to point to the lack of `CMN_BASS.ZDL` and `BCAB` references in the firmware. The CAB processing is probably split off into `CMN_BASS.ZDL`. This is also different from how it's done with the MS-50G Guitar CABs, where the choices correspond to the AMP effects actually on the pedal.

The "underscored" effects are migrated from B1on. Even though mostly they are the same, some parts (the CAB handing) differ; also the MS-60B AMP ZDL are smaller. The sounding appears mostly identical.