Barsik-Barbosik/Zoom-Firmware-Editor #12: effect list

randysampang, 2019-08-06:
Thank you so much for this program. do you happen to have a side by side list of effects for each model? i remember seeing an excel type list but can't seem to find it anywhere. thanks in advance form the Philippines!

hardrubbish, 2019-08-06:
I think I saw it linked in a forum the other day... 
I'd be interested to see the difference in size of the effects across different pedals.
I don't think they're 1:1 copies. I recall people going deep into A/Bing the matching effects on the MS50G expanded effect FW update and the MS100BT. Apparently they sounded different.
I'll post the list if I see it again.

hardrubbish, 2019-08-06:
Here's the MultiStomp list.
https://www.reddit.com/r/zoommultistomp/comments/7vqqsw/alphabetical_list_of_all_effects_ms70cdrms60bms50g/

Barsik-Barbosik, 2019-08-06:
Later I will create a similar list of all possible effects with their descriptions. I'm going to show the effect description (tooltip or new column), but there are no embedded descriptions inside the effect files. So, the external list is required.

hardrubbish, 2019-08-06:
Awesome!
It’d be interesting to see the file/block sizes all laid out too.

randysampang, 2019-08-08:
Found it! 

https://docs.google.com/spreadsheets/d/1ng6M2bq-QMP38MCeqbCo-6RjLPA9mrzHUCh8yDtdYNE/edit?usp=sharing

hardrubbish, 2019-08-08:
Nice work!!

randysampang, 2019-08-08:
it doesn't have the list for the B1XOn though.

mungewell, 2019-08-22:
Here's an auto-magically made list of the 'ZD2' effects on the G3n and friends.... based on the contents of the FLST_SEQ.ZT2 files.
[effect_summary.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3528207/effect_summary.zip)

Suggestions on how to improve welcomed.