Barsik-Barbosik/Zoom-Firmware-Editor #14: Adding effects to G1Four - failed

mungewell, 2019-08-20:
Hi,
I attempted to add 2 SFX effects from the B1Four to the G1Four f/w image. I uploaded the 'ZD2' files at the end and removed then added a modified 'FLST_SEQ.ZT2' (python script here: 
[zoom_zt2.py.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3518663/zoom_zt2.py.txt) )
```
C:\Users\simon\Documents\ZoomFW>"c:\Program Files\Python36\python.exe" zoom_zt2.py -w -A 7 -n "Z_SYN_1U.ZD2" -v "1.00" -i 49 -u 128 G1_FOUR_v1.10_Win_E\mod_20190819\FLST_SEQ.ZT2
Reading G1_FOUR_v1.10_Win_E\mod_20190819\FLST_SEQ.ZT2...
Group 7:
    BOMBER.ZD2
    AUTOPAN.ZD2
    LPROLL3S.ZD2
    HOTSPICE.ZD2
Add:Z_SYN_1U.ZD2

C:\Users\simon\Documents\ZoomFW>"c:\Program Files\Python36\python.exe" zoom_zt2.py -w -A 7 -n "DEFRET.ZD2" -v "1.20" -i 64 -u 128 G1_FOUR_v1.10_Win_E\mod_20190819\FLST_SEQ.ZT2
Reading G1_FOUR_v1.10_Win_E\mod_20190819\FLST_SEQ.ZT2...
Group 7:
    BOMBER.ZD2
    AUTOPAN.ZD2
    LPROLL3S.ZD2
    HOTSPICE.ZD2
    Z_SYN_1U.ZD2
Add:DEFRET.ZD2
```
The f/w image download OK, but the effects are not visible on the pedal's menus, or in Guitar Lab.

Other issues talk about a 'CMN_BASS.ZDL', this (or 'ZD2' equivalent) is not listed in the B1Four F/W.

hardrubbish, 2019-08-20:
Is there a CMN_DRV.ZDL file? I needed that for Multistomp, for some guitar effects.
I found that g200kg and ToneLib editors didn't display the injected effects, even once the pedal did. I think they're expecting only a limited range of effects from each pedal.

mungewell, 2019-08-20:
There are no "CMN_" files listed in the f/w image at all.

One interesting thing is that the pedal is listing the effects installed/removed from the previous f/w (shipped as ver 1,00). I had removed almost all amps/cabs to make space for over loadable effects.

I did try a factory reset, but listings remain the same.

mungewell, 2019-08-20:
It's almost as if it is ignoring my 'FLST_SEQ.ZT2' file and using one from previous Guitar Lab session.

I tried flashing the official V1.10 image, but it still doesn't list all the amps/cabs which that image contains....

hardrubbish, 2019-08-20:
Oh, that sounds weird. So you think that GuitarLab is referencing a previous version of the pedal's FW? 
Sounds like these programs all get a bit confused when there's effects that they aren't programmed to expect.

mungewell, 2019-08-20:
Found an error with my python script in the header it produced and ensured that the file as exactly the same length (padding zero) as other f/w. Still no joy. Modified file attached for reference.
[FLST_SEQ.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3518920/FLST_SEQ.zip)

The pedal itself (not just Guitar Lab) is not listing the Amp/Cab effects which I had previously deleted (v1.00) and _are_ included in v1.10 - both as ZDL and within the 'FLST_SEQ.ZT2'.

Barsik-Barbosik, 2019-08-22:
Have you tried to insert new effect names into GUARDZDL.ZT2?

mungewell, 2019-08-22:
I did not adjust 'GUARDZDL.ZT2', I am 99% convinced this is a list of effects NOT to use on the G1/B1Four.... specifically if you look it gives the (old) names for effects which are replaced with the '1U' or '3S' variants. 

See #6 for details.

Barsik-Barbosik, 2019-08-22:
You are right, GUARDZDL.ZT2 consists of old effect names.

> It's almost as if it is ignoring my 'FLST_SEQ.ZT2' file and using one from previous Guitar Lab session.

Maybe, it will update the effect list if you "restore factory default settings" (turn the device power on, while pressing "memory/stomp/edit" button, and accept initialization)? But I'm not sure.

Barsik-Barbosik, 2019-08-22:
One more possible test: try to add previous version (1.10) of DEFRET.ZT2 (from G3n v2.00 firmware).

And I just found that FLST_SEQ.ZT2 from G1X FOUR has extra elements that are missing in the firmware. For example: THEVIBE.ZD2, LPROLL3S.ZD2. Does that list contains all possible effects for that pedal, that are available in Guitar Lab?

mungewell, 2019-08-22:
I did try the factory reset, but have spotted/made minor corrections to the script I was using. I will have a 'do over' and attempt again from scratch. Maybe with some other effects.

Checkout the zip I added to #12. The Defret from B3n (v2.20) is exactly the same as the B1Four. I will see if I can find B3n (v2.00)
```
7d2b2ca0973aed4634f39799a09aaead | ./B3n_v2.20_Win_E/effects/DEFRET.ZD2
7d2b2ca0973aed4634f39799a09aaead | ./B1_FOUR_v1.10_Win_E/effects/DEFRET.ZD2
```

mungewell, 2019-08-22:
Some hints on how effects are deactivated here:
https://github.com/PedalController/MidiMultistompController/blob/master/testmidolist2.py#L237
```
# Ativar efeito - Comando grande
# Desativar - F0 52 00 5A 31 [effect] 00 00 00 F7
# Trocar efeito - Comando grande
# Trocar de banco - Comando Pequeno (CC 00, CC 32, PC [patch number])
# Alterar parâmetro - Comando Grande
```
What is that 'Comando grande' though.....

mungewell, 2019-08-23:
Tried the 'Do Over', and failed again.... very careful about getting everything right.
Ran the uploader (on Win10 64bit) and that appeared to flash the pedal.Is there any way to confirm that the download process (really) succeeded, other than the GUI on PC/Pedal.
Menu on the pedal shows the 'removed' effects and not the 'new' bass ones.

Using SysEx I attempted to select the 'new' effect via their ID's, but got shown the '/!\' Warning triangle (Missing Effect Not Found).
`amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 01 31 00 00 3C 00 F7'`

mungewell, 2019-09-07:
So excited to share this image... uploaded via Midi, full explanation later.
[image]

mungewell, 2019-09-08:
see #16

ZinnatBass, 2019-10-14:
@mungewell 
Thank you very much for your fantastic work on the G1 four. I would like to confirm that, as you were expecting, your code works like a charm also on the B1 four. I was able to load and test 5 amp sims from the G1 four firmware.

However, I later encountered a small issue which I would like to mention. After installing and testing the 5 additional guitar amp sims, ~98% of the memory was used. I then added a cab sims from the G1 four firmware. Something went wrong with the new listing in the FLST_SEQ.ZT2: most probably the memory was already completely filled up by the upload of the new effect binaries when it was time to modify the FLST_SEQ.ZT2. The consequence was that most of the original B1 four effects and the 5 additional guitar amps disappeared, despite the presence of the .ZD2 files (due to a wrong checksum, as detected also by your GUI). Interestingly the cab sim added last was present.

I then reinitialized to factory conditions, uploaded a fresh original firmware v1.1, and re-uploaded the original B1 four effects with the Zoom Guitar Lab official application. I used your GUI to check the content of the memory and figured that the additional .ZD2 files of the 5 guitar amps and of the 1 guitar cab sim were still present in the memory. I then used your "-U" command to remove them.

At that point, I wanted to upload again at least the 5 amp sims from the G1 four (skipping the cab sim which cause the issue), but I ran almost out of memory after installing only 3 of them this time.

My guess is that there could be still in the memory some remainder file from the above incident, despite I don't see anything wrong in the file list extracted by your GUI. I read that at some point you removed everything from your memory and noticed 22% of it was still used: did you figure out how to clean it up completely?

It would be fantastic if you could add a command to list the entire memory content, to try to sort also this issue out. I will mention this suggestion also on your repository.

Thank you very much once again for this incredible work! It's admirable what you were able to achieve.

mungewell, 2019-10-14:
Thank you for confirming the B1Four functions, I also see that you have discovered the 'sister' project.

ZinnatBass, 2019-10-16:
Thank you for the answer, Mungewell! I think I have figured out the reason why my memory usage didn't seem to add up: most probably it was because when I messed up the 'FLST_SEQ.ZT2' I used Zoom Guitar Lab to inject from scratch the stock B1 effects, and that caused to install 1 or 2 extra effects which Zoom have released meanwhile for the B1, but were not available on my stock firmware. The 'files' tab in your GUI (sister project) does indeed list the entire memory content. Sorry for the confusion. Thanks!

clint-toy, 2019-12-19:
@ZinnatBass  how did you manage to install effects from G1 Four to B1 Four?  I used the gui tool by @mungewell however, it says checksum error after the Install function.

@mungewll looks like the Zoom Guitar Lab downloads and keeps a copy of the files (including FLST) library in the host pc (in user/appdata/local/zoom folder) which maybe being synced to the pedal all the time.

I would like to install a few guitar amp sims to extend my sound library and a help is greatly appreciated.  By the way my device is now at 2.0 firmware.

clint-toy, 2019-12-25:
This is resolved by using the Zoom Guitar Lab application to unload/delete the effects (they remain in the effects list like inactive status) and remove other effects to make room. Then, add/install again the desired effects into the pedal. Thanks!!!

I think the zoomzt2 tool needs to have a flag if the device memory is already full or unable to load effects due to memory capacity.

githubguitargeek, 2020-06-04:
Any windows utility to use this as a windows.exe. A lot of users like myself would like to add extra effects but have no experience in running python guis. It will be great if we have a windows utility to run this.

mungewell, 2020-06-04:
@githubguitargeek There is a EXE in the Zip file of the 'Release' tab...
https://github.com/mungewell/zoom-zt2/releases

githubguitargeek, 2020-06-07:
Thanks for the link mungewell. Appreciate your help. Few effects don't load in to the G1XFOUR firmware like the Particle Reverb and HOLD Delay. I will make a list and upload all the effects that won't load in to G1XFour

yadu035, 2020-09-10:
Will the codes made for G1four work in G1xfour. Do I need to change anything to install effects on my G1xfour

jemdl1998, 2020-12-15:
> Why is it all my electronics fall apart? ;-)
> [image]
> 
> IC3 is a SPI EEPROM:
> https://www.macronix.com/Lists/Datasheet/Attachments/7351/MX25L3273E,%203V,%2032Mb,%20v1.1.pdf
> 
> and is connected to the CN2 footprint (and TPs) as follows
> 
> ```
> CN2		IC3
> 1	TP1	8	VCC
> 2	TP2	6	SCLK
> 3	TP3	2	S0/SIO1
> 4	TP4	5	SI/SIO0
> 5	-	-
> 6	TP5	1	CS#
> 7	TP6	4	GND
> ```
> 
> Who's betting that it contains the current settings (installed/de-installed effects)?

Do you know where the expression pedal of the g1xfour would go? Or don't they use the same board?