Barsik-Barbosik/Zoom-Firmware-Editor #15: Mapping of Effect ID: FLST_SEQ <-> SysEx

mungewell, 2019-08-23:
Getting familar with  SysEx control of the pedals, it seems that the command to place effects into the various 'slots' uses a ID as follows.
https://www.thegearpage.net/board/index.php?threads/midi-control-of-zoom-g3-video.1033719/page-3
```
Selecting effect from the editor:
selecting Comp in slot 6 sends F0 52 00 5A 31 05 01 17 00 F7,
selecting MComp in slot 6 sends F0 52 00 5A 31 05 01 19 00 F7,
selecting ZNR: F0 52 00 5A 31 05 01 1B 00 F7
selecting Pedal pitch: F0 52 00 5A 31 00 01 45 00 F7

So the 8th byte is the effect type.
This is consistent with the effect type list below: 17hex = 23 dec
```
```
Table: effect types - numbers are decimal:
000, M-FILTER
001, THE VIBE
002, Z-ORGAN
003, PEDAL VX
004, PHASE DELAY
005, FILTER DELAY
... 
```
Do 'we' know if/how these relate to the placement of the effects in FLST_SEQ?

mungewell, 2019-08-23:
Another reference for the Effect IDs over SysEx
https://github.com/sixeight7/VController_v2/blob/master/VController_v2/MIDI_ZG3.ino#L324

mungewell, 2019-08-23:
So, that page is mostly correct but I found that some of the commands are different for the G1Four.
```
Turn on/off effect slots
--
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 00 00 01 00 00 00 00 F7' - slot 1, on
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 00 01 00 00 00 00 F7' - slot 2, on
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 00 00 00 00 00 00 F7' - slot 2, off
etc..
```
Finding out how to set which effect is used in a slot was a real pain, they've done some weird bit shifting.
```
Bypass Slot 2
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 01 00 00 00 00 00 F7'

Load Effect into Slot 2 - ! = unknown effect (haven't found a valid one yet)
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 01 00 50 00 00 00 F7'

load "rack comp" to slot 1
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 00 01 20 00 00 08 00 F7'

load "slow attack" to slot 1
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 00 01 30 00 00 08 00 F7'
                                           ^^    ^^       ^^- Group??? should be group 1 << 3bits!!
                                           ||    ++- Effect (ID from FLST_SEQ)
                                           ++- Slot

load "rack comp" to slot 2
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 01 20 00 00 08 00 F7'
EP_STOMP.ZD2 (ver= 1.30 ), group= 3 , id= 32 , installed= 1

load "EP Stamp" to slot 2
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 01 20 00 00 18 00 F7'
EP_STOMP.ZD2 (ver= 1.30 ), group= 3 , id= 32 , installed= 1

load "MS2x14" to slot 2
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 01 10 00 00 28 00 F7'
MS4X12.ZD2 (ver= 1.20 ), group= 5 , id= 16 , installed= 1

load "Bomber" to slot 2
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 01 10 00 00 38 00 F7'
BOMBER.ZD2 (ver= 1.30 ), group= 7 , id= 16 , installed= 1

load "blackOpt" to slot 2
amidi -p hw:1,0,0 -S 'F0 52 00 6E 64 03 00 01 01 10 00 00 0C 00 F7'
BLACKOPT.ZD2 (ver= 1.20 ), group= 1 , id= 16 , installed= 0, unknown=1
```

shooking, 2020-12-09:
hi - wonderful stuff you are all doing.
I got a Zoom B1XFour and G1XFour recently.
Started down the Wireshark / parse sysex sentences then cut and paste into MidiOx.
Anyhow thanks for all you done and if I may what is the best way to contribute some extra(?) info?

For example, I too was stuck on the CMD 31 for g3 v G1X.
I could see 64 was the magic string based on USB decoding - glad I stumbled across your work.

So "bypass" of 64 is a special case of setting the FX to 0 right.
But do you know it is possible to get the Zoom to tell you current status of FX?

pi@raspberrypi:~/Software/ZoomPedal $ theCMD=64; amidi -p ${MIDI_DEV} -S "F0 52 00 6E $theCMD 02 bp ep 00 F7" -r t1.bin -t 1; hexdump -C t1.bin; rm t1.bin

Here bp is beginning parameter (param - 1 as ever)
ep is ending parameter (" ")
ONE caveat - if an FX takes up 2 slots (like a Fender Twin) then you need to look a m. m+1 for slot m and all other slots are moved up by one.

EXAMPLE
Let's move to patch Randy Train (why not)

pi@raspberrypi:~/Software/ZoomPedal $ theCMD=64; amidi -p ${MIDI_DEV} -S "F0 52 00 6E $theCMD 02 00 08 00 F7" -r t1.bin -t 1; hexdump -C t1.bin; rm t1.bin
Show me ASCII for params 0 ... 8 (I guess the most I ever seen is 8 or 9)

pi@raspberrypi:~/Software/ZoomPedal $ theCMD=64; amidi -p ${MIDI_DEV} -S "F0 52 00 6E $theCMD 02 00 08 00 F7" -r t1.bin -t 1; hexdump -C t1.bin; rm t1.bin

1023 bytes read
00000000  f0 52 00 6e 64 01 00 00  00 00 31 00 00 00 00 00  |.R.nd.....1.....|
00000010  00 00 00 00 00 00 01 00  4f 6e 4f 66 66 00 00 00  |........OnOff...|
00000020  00 00 00 01 00 00 5a 4e  52 00 00 00 00 00 00 00  |......ZNR.......|
00000030  00 01 01 00 5a 4e 52 00  00 00 00 00 00 00 00 02  |....ZNR.........|
00000040  00 00 45 46 58 49 4e 00  00 00 00 00 00 02 01 00  |..EFXIN.........|
00000050  44 45 54 43 54 00 00 00  00 00 00 03 00 00 31 30  |DETCT.........10|
00000060  30 00 00 00 00 00 00 00  00 03 01 00 44 65 70 74  |0...........Dept|
00000070  68 00 00 00 00 00 00 04  00 00 33 30 00 00 00 00  |h.........30....|
00000080  00 00 00 00 00 04 01 00  54 48 52 53 48 00 00 00  |........THRSH...|
00000090  00 00 00 05 00 00 30 00  00 00 00 00 00 00 00 00  |......0.........|
000000a0  00 05 01 00 44 65 63 61  79 00 00 00 00 00 01 00  |....Decay.......|
000000b0  00 00 31 00 00 00 00 00  00 00 00 00 01 00 01 00  |..1.............|
000000c0  4f 6e 4f 66 66 00 00 00  00 00 01 01 00 00 50 61  |OnOff.........Pa|
000000d0  72 61 45 51 00 00 00 00  01 01 01 00 50 61 72 61  |raEQ........Para|
000000e0  45 51 00 00 00 00 01 02  00 00 20 38 30 30 20 00  |EQ........ 800 .|
000000f0  00 00 00 00 01 02 01 00  46 52 45 51 00 00 00 00  |........FREQ....|
00000100  00 00 01 03 00 00 20 32  2e 30 20 00 00 00 00 00  |...... 2.0 .....|
00000110  01 03 01 00 51 00 00 00  00 00 00 00 00 00 01 04  |....Q...........|
00000120  00 00 38 2e 35 00 00 00  00 00 00 00 01 04 01 00  |..8.5...........|
00000130  47 61 69 6e 00 00 00 00  00 00 01 05 00 00 38 35  |Gain..........85|
00000140  00 00 00 00 00 00 00 00  01 05 01 00 56 4f 4c 00  |............VOL.|
00000150  00 00 00 00 00 00 02 00  00 00 31 00 00 00 00 00  |..........1.....|
00000160  00 00 00 00 02 00 01 00  4f 6e 4f 66 66 00 00 00  |........OnOff...|
00000170  00 00 02 01 00 00 44 49  53 54 20 50 6c 75 73 00  |......DIST Plus.|
00000180  02 01 01 00 44 49 53 54  20 50 6c 75 73 00 02 02  |....DIST Plus...|
00000190  00 00 31 30 30 00 00 00  00 00 00 00 02 02 01 00  |..100...........|
000001a0  47 61 69 6e 00 00 00 00  00 00 02 03 00 00 31 30  |Gain..........10|
000001b0  30 00 00 00 00 00 00 00  02 03 01 00 56 4f 4c 00  |0...........VOL.|
000001c0  00 00 00 00 00 00 02 04  00 00 30 00 00 00 00 00  |..........0.....|
000001d0  00 00 00 00 02 04 01 00  44 72 79 4d 78 00 00 00  |........DryMx...|
000001e0  00 00 02 05 00 00 4f 52  47 00 00 00 00 00 00 00  |......ORG.......|
000001f0  02 05 01 00 43 6f 6d 70  00 00 00 00 00 00 03 00  |....Comp........|
00000200  00 00 31 00 00 00 00 00  00 00 00 00 03 00 01 00  |..1.............|
00000210  4f 6e 4f 66 66 00 00 00  00 00 03 01 00 00 47 74  |OnOff.........Gt|
00000220  20 47 45 51 20 37 00 00  03 01 01 00 47 74 20 47  | GEQ 7......Gt G|
00000230  45 51 20 37 00 00 03 02  00 00 37 2e 30 00 00 00  |EQ 7......7.0...|
00000240  00 00 00 00 03 02 01 00  31 30 30 00 00 00 00 00  |........100.....|
00000250  00 00 03 03 00 00 32 2e  30 00 00 00 00 00 00 00  |......2.0.......|
00000260  03 03 01 00 32 30 30 00  00 00 00 00 00 00 03 04  |....200.........|
00000270  00 00 33 2e 30 00 00 00  00 00 00 00 03 04 01 00  |..3.0...........|
00000280  34 30 30 00 00 00 00 00  00 00 03 05 00 00 32 2e  |400...........2.|
00000290  30 00 00 00 00 00 00 00  03 05 01 00 38 30 30 00  |0...........800.|
000002a0  00 00 00 00 00 00 04 00  00 00 30 00 00 00 00 00  |..........0.....|
000002b0  00 00 00 00 04 00 01 00  44 75 6d 6d 79 00 00 00  |........Dummy...|
[... 124 lines cut ...]
00000060  30 00 00 00 00 00 00 00  03 03 01 00 32 30 30 00  |0...........200.|
00000070  00 00 00 00 00 00 03 04  00 00 33 2e 30 00 00 00  |..........3.0...|
00000080  00 00 00 00 03 04 01 00  34 30 30 00 00 00 00 00  |........400.....|
00000090  00 00 03 05 00 00 32 2e  30 00 00 00 00 00 00 00  |......2.0.......|
000000a0  03 05 01 00 38 30 30 00  00 00 00 00 00 00 04 00  |....800.........|
000000b0  00 00 30 00 00 00 00 00  00 00 00 00 04 00 01 00  |..0.............|
000000c0  44 75 6d 6d 79 00 00 00  00 00 04 01 00 00 47 74  |Dummy.........Gt|
[... cut ...]
00000150  00 00 00 00 00 00 f7                              |.......|
00000157

And that seems to cover from 100 == 7.0 pg1 to VOL 74 page 3

You get the idea.

Anyhow I am able to read and set parameters, read FX. Working on your issue above to set FX. Thanks for all the hints collectively give thus far.

I think I will start looking at the Firmware - I used to have the Miniak firmware - Alesis/AKAI never published that - managed to crack 3 of the 4 format - multi's was the PITA. Patches (was given), Sequences and Rhythms was fun ;-).

shooking, 2020-12-09:
Regarding the "slots" - I can see this has something to do with some index based I guess on the firmware (and related to the order in which the G1X reads stuff)

Getting late here but I found this interesting - hoping it means more to you OR you already discovered this and I not learnt where to look
pi@raspberrypi:~/Software/ZoomPedal $ theCMD=64; amidi -p ${MIDI_DEV} -S "F0 52 00 6E $theCMD 05 F7" -r t1.bin -t 1; hexdump -C t1.bin; rm t1.bin

115 bytes read
00000000  f0 52 00 6e 64 06 52 41  43 4b 43 4f 4d 50 2e 5a  |.R.nd.RACKCOMP.Z|
00000010  44 32 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |D2..............|
00000020  00 00 00 00 08 01 00 00  00 00 00 00 00 00 00 00  |................|
00000030  32 00 28 00 00 00 10 00  00 00 00 09 00 05 00 00  |2.(.............|
00000040  00 00 00 00 00 00 09 00  06 00 00 00 00 00 00 00  |................|
00000050  00 64 00 50 00 00 00 16  00 00 00 00 00 00 00 00  |.d.P............|
00000060  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000070  00 00 f7                                          |...|
00000073
pi@raspberrypi:~/Software/ZoomPedal $ theCMD=64; amidi -p ${MIDI_DEV} -S "F0 52 00 6E $theCMD 05 00 00 00 00 00 00 00 00 F7" -r t1.bin -t 1; hexdump -C t1.bin; rm t1.bin

93 bytes read
00000000  f0 52 00 6e 64 06 53 4c  57 41 54 4b 2e 5a 44 32  |.R.nd.SLWATK.ZD2|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  00 00 00 00 06 01 00 00  00 00 00 00 00 00 00 00  |................|
00000030  31 00 14 00 00 00 10 00  00 00 00 0a 00 0a 00 00  |1...............|
00000040  00 10 00 00 00 00 64 00  64 00 00 00 00 00 00 00  |......d.d.......|
00000050  00 64 00 50 00 00 00 16  00 00 00 00 f7           |.d.P.........|
0000005d

so as I added more 0's on I can see I am getting the file names?
In terms of this current patch
FX1 = SlowATTCK
FX2 = ZNR
FX3 = FD TWNR ... I think this takes 2 slots?
FX4/5 = Room
FX5/6 = AutoWah

so 64 05 seems to be something related to files?

Actually

pi@raspberrypi:~/Software/ZoomPedal $ theCMD=64; amidi -p ${MIDI_DEV} -S "F0 52 00 6E $theCMD 05 00 F7" -r t1.bin -t 1; hexdump -C t1.bin; rm t1.bin

49 bytes read
00000000  f0 52 00 6e 64 06 42 59  50 41 53 53 2e 5a 44 32  |.R.nd.BYPASS.ZD2|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  00 00 00 00 02 01 00 00  00 00 00 00 00 00 00 00  |................|
00000030  f7                                                |.|
00000031

And I was on bypass

pi@raspberrypi:~/Software/ZoomPedal $ theCMD=64; amidi -p ${MIDI_DEV} -S "F0 52 00 6E $theCMD 05 01 F7" -r t1.bin -t 1; hexdump -C t1.bin; rm t1.bin

93 bytes read
00000000  f0 52 00 6e 64 06 43 4f  4d 50 2e 5a 44 32 00 00  |.R.nd.COMP.ZD2..|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  00 00 00 00 06 01 00 00  00 00 00 00 00 00 00 00  |................|
00000030  0a 00 06 00 00 00 10 00  00 00 00 01 00 00 00 00  |................|
00000040  00 00 00 00 00 00 0a 00  06 00 00 00 00 00 00 00  |................|
00000050  00 64 00 50 00 00 00 16  00 00 00 00 f7           |.d.P.........|
0000005d

And now FX1 is on COMP

but clearly more params are required because more often than not this crashes the G1XFour ... will sleep on it.

mungewell, 2020-12-09:
@shooking 

I think you'd be real interested in the sister project which is targeted at the G1Four (and friends):
https://github.com/mungewell/zoom-zt2

I had been making notes about the detailed Midi control of the device(s), but I had my pedal die earlier in the year. You seem like you are making great progress, and I would like to collaborate if you want... I can share my notes with you too, but lets open bugs over here...
https://github.com/mungewell/zoom-zt2/issues

[midi_notes.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/5662857/midi_notes.txt)

shooking, 2021-01-01:
Firstly wishing you a great new year - hopefully a lot better than the last one :-(

Hi so I was stuck on processing the Zoom output with my Raspberry PI Ctrlr and Lua.
Short version ... I have an alternative way of interpreting the FXID, GID from a ZT2 dump

```
LMT1176.ZD2
1.50
01 20
80 01
AHH these are 8 bit because converts from 8 midi byte to 7
And it is an endianess?
so 80 01 is really 180(h)
for some reason we right shift 5 to get the GID/

So 180(h) >>5 = 0c
   100(h) >>5 = 08
SL = slot 0 .. 8
L1/H1 = low/high value of FX ID (8 bit value)
GS= GID (01 80, 01 00) >> 5

 amidi -p ${MIDI_DEV} -S "F0 52 00 6E 64 03 00 SL 01 L1 H1 00 GS 00 F7" -r pants.bin -t1; hexdump -C pants.bin; rm pants.bin

```

Long version.

Linux (well RPi at least) ALSA MIDI sysex seems limited to 256 bytes.
I was expecting 603 but always came up 256 short ... this lead me to reinterpret data and I see the GroupID shift different to you now.
Anyhow I realised some of the file capture commands have a block size. Changed from 0x04 to 0x02 and now I get "256" chunks rather than "512".
I hacked at your zoomzt2.py, modified that to accept 256 rather than 512 and now I am am able to grab the ZT2's into LUA at least.

So what? Well I started to unpick what you did with Construct. That was too smart for me ... so I played and now I see the format differently to you - I suspect it amounts to the same thing .. but the swines just ASCII coded things right??

So your decoded comes out with strings like:

```
dump

ListContainer: 
    Container: 
        a = b'>>>\x00' (total 4)
        b = None
        name = u'BYPASS.ZD2' (total 10)
        c = None
        d = b'1' (total 1)
        e = None
        f = b'<<<\x00' (total 4)
        g = None
    ListContainer: 
        Container: 
            group = 1
            groupname = (enum) DYNAMICS 1
            effects = ListContainer: 
                Container: 
                    effect = u'COMP.ZD2' (total 8)
                    version = u'1.40' (total 4)
                    installed = 1
                    id = 16777232
                    group = 1
                Container: 
                    effect = u'RACKCOMP.ZD2' (total 12)
                    version = u'1.40' (total 4)
                    installed = 1
                    id = 16777248
                    group = 1
                Container: 
                    effect = u'SLWATK.ZD2' (total 10)
                    version = u'1.50' (total 4)
                    installed = 1
[... 325 lines cut ...]
1.30
01 10  16 1 dont work. Notice we ahve 80. What do do?
80 01

81 x 8 ??
LMT1176.ZD2
1.50
01 20
80 01
AHH these are 8 bit because converts from 8 midi byte to 7
And it is an endianess?
so 80 01 is really 180(h)
for some reason we right shift 5 to get the GID/

So 180(h) >>5 = 0c
   100(h) >>5 = 08
160_COMP.ZD2
1.30
01 30
80 01
DUAL_CMP.ZD2

Bloody big v little endian -- I missed out 16 bit and jumped to 32 -- always liked Motorola better than IBM.
So I was biased and looking at the data the wrong way.
But now I cannot unsee it -- ASCII and big endian?

shooking, 2021-01-01:
So I recoded to swap bytes

```
BYPASS.ZD2

FXID: 00 int: 0
GID: 0000 int: 0
SLWATK.ZD2
1.40
FXID: 30 int: 48
GID: 0001 int: 256
ZNR.ZD2
1.40
FXID: 40 int: 64
GID: 0001 int: 256
OPTCOMP.ZD2
1.50
FXID: 80 int: 128
GID: 0001 int: 256
BLACKOPT.ZD2
1.30
FXID: 10 int: 16
GID: 8001 int: 384
LMT1176.ZD2
1.50
FXID: 20 int: 32
GID: 8001 int: 384
160_COMP.ZD2
1.30
FXID: 30 int: 48
GID: 8001 int: 384
DUAL_CMP.ZD2
1.30
FXID: 40 int: 64
GID: 8001 int: 384
MB_COMP.ZD2
1.30
FXID: 50 int: 80
GID: 8001 int: 384
DCOMP.ZD2
1.30
FXID: 60 int: 96
GID: 8001 int: 384
GLAMCOMP.ZD2
1.20
FXID: 70 int: 112
GID: 8001 int: 384
SEQFLTR.ZD2
1.40
FXID: 40 int: 64
GID: 0002 int: 512
```

So now I can do things like

pi@raspberrypi:~/Software/ZoomPedal $ ./FXM_ID.sh 1 32 640

pi@raspberrypi:~/Software/ZoomPedal $ ./FX_M1_M2_Values.sh 1

|Z Tron........ 7|

pi@raspberrypi:~/Software/ZoomPedal $ ./FXM_PN.sh 1 1 0

pi@raspberrypi:~/Software/ZoomPedal $ ./FX_M1_M2_Values.sh 1

....Z Tron........-10...........|
pi@raspberrypi:~/Software/ZoomPedal $ ./FXM_PN.sh 1 1 1024
pi@raspberrypi:~/Software/ZoomPedal $ ./FX_M1_M2_Values.sh 1
pi@raspberrypi:~/Software/ZoomPedal $ ./GetMoreData.sh  # flush the buffer
pi@raspberrypi:~/Software/ZoomPedal $ ./FX_M1_M2_Values.sh 1
|....Z Tron........10............|

So by introspection we can see the 1st param of patch Z Tron midi [0, 1024] -> [-10, 10] so likely [0, 21] -> [-10, 10]

and repeating similar shows this to be the case.

Nice. I feel some Ctrlr panel comlng along.