Barsik-Barbosik/Zoom-Firmware-Editor #16: G1Four - reading FLST_SEQ via Midi

mungewell, 2019-09-01:
Seems that my suspicion was correct, the G1Four (and presumably other Series2 pedals) maintain a 'FLST_SEQ' file outside of the F/W and this is communicated back/forth over Midi as effects are installed and remove.

Poking around I was able to read (from pedal to PC) the following
```
$ amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r prog.bin -t 1 ; hexdump -C prog.bin
  

603 bytes read
00000000  f0 52 00 6e 60 04 22 01  00 00 04 00 3e 3e 3e 00  |.R.n`.".....>>>.|
00000010  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000020  00 00 00 00 00 00 00 00  00 42 59 00 50 41 53 53  |.........BY.PASS|
00000030  2e 5a 44 00 32 00 00 00  00 00 00 00 00 00 01 00  |.ZD.2...........|
00000040  00 00 00 00 00 00 00 3c  3c 3c 00 00 00 00 00 00  |.......<<<......|
00000050  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000060  00 00 00 00 00 3e 3e 3e  00 01 00 00 00 00 00 00  |.....>>>........|
00000070  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000080  00 00 43 00 4f 4d 50 2e  5a 44 32 00 00 00 00 00  |..C.OMP.ZD2.....|
00000090  00 31 2e 00 32 30 00 00  10 00 00 00 01 00 00 00  |.1..20..........|
000000a0  52 41 43 00 4b 43 4f 4d  50 2e 5a 00 44 32 00 31  |RAC.KCOMP.Z.D2.1|
000000b0  2e 32 30 00 00 00 20 00  00 01 00 00 00 00 53 4c  |.20... .......SL|
000000c0  57 41 54 00 4b 2e 5a 44  32 00 00 00 00 31 2e 33  |WAT.K.ZD2....1.3|
000000d0  30 00 00 00 30 00 00 01  00 00 00 00 5a 4e 52 2e  |0...0.......ZNR.|
000000e0  5a 44 32 00 00 00 00 00  00 00 31 00 2e 33 30 00  |ZD2.......1..30.|
000000f0  00 40 00 00 00 01 00 00  00 4d 55 00 54 45 53 57  |.@.......MU.TESW|
00000100  2e 5a 44 00 32 00 00 00  31 2e 32 00 30 00 00 50  |.ZD.2...1.2.0..P|
00000110  00 00 01 00 00 00 00 47  52 41 59 00 43 4f 4d 50  |.......GRAY.COMP|
00000120  2e 5a 44 00 32 00 31 2e  32 30 00 00 00 60 00 00  |.ZD.2.1.20...`..|
00000130  01 00 00 00 00 4e 5f 47  41 54 45 00 2e 5a 44 32  |.....N_GATE..ZD2|
00000140  00 00 00 00 31 2e 32 30  00 00 70 00 00 00 01 00  |....1.20..p.....|
00000150  00 00 4f 00 50 54 43 4f  4d 50 2e 00 5a 44 32 00  |..O.PTCOMP..ZD2.|
00000160  00 31 2e 04 33 30 00 00  00 00 00 00 01 00 00 00  |.1..30..........|
00000170  42 4c 41 00 43 4b 4f 50  54 2e 5a 00 44 32 00 31  |BLA.CKOPT.Z.D2.1|
00000180  2e 32 30 04 00 01 10 00  00 01 00 00 00 00 4c 4d  |.20...........LM|
00000190  54 31 31 00 37 36 2e 5a  44 32 00 00 00 31 2e 33  |T11.76.ZD2...1.3|
000001a0  30 00 00 10 20 00 00 01  00 00 00 00 3c 3c 3c 00  |0... .......<<<.|
000001b0  01 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000001c0  00 00 00 00 00 00 00 00  00 3e 3e 00 3e 00 02 00  |.........>>.>...|
000001d0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
000001e0  00 00 00 00 00 00 00 41  55 54 4f 00 57 41 48 2e  |.......AUTO.WAH.|
000001f0  5a 44 32 00 00 00 31 2e  32 30 00 00 00 10 00 00  |ZD2...1.20......|
00000200  02 00 00 00 00 52 45 53  4f 4e 41 00 4e 43 2e 5a  |.....RESONA.NC.Z|
00000210  44 32 00 00 31 2e 32 30  00 00 20 00 00 00 02 00  |D2..1.20.. .....|
00000220  00 00 43 00 52 59 2e 5a  44 32 00 00 00 00 00 00  |..C.RY.ZD2......|
00000230  00 31 2e 00 32 30 00 00  30 00 00 00 02 00 00 00  |.1..20..0.......|
00000240  53 45 51 00 46 4c 54 52  2e 5a 44 00 32 00 00 31  |SEQ.FLTR.ZD.2..1|
00000250  2e 33 30 00 00 5a 50 56  56 06 f7                 |.30..ZPVV..| 
```

Not 100% sure of the sequence of 'commands' required, as I only read it once.... but adjusted my Python script to decode this - it has the 8byte->7byte decoding for filling in the high-bits.

[reading_flst_seq.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3563758/reading_flst_seq.zip)

mungewell, 2019-09-02:
The FLST_SEQ is read in multiple parts (17 for G1Four, maybe different for others).
Note: the '6E' is for G1Four, other pedals will need different SysEx ID.

```
amidi -p hw:1,0,0 -S "f0 52 00 6e 52 f7" -r temp.bin -t 1 ; hexdump -C temp.bin

amidi -p hw:1,0,0 -S "f0 52 00 6e 60 25 00 00 46 4c 53 54 5f 53 45 51 2e 5a 54 32 00 05 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 27 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 20 02 00 00 00 00 00 00 00 00 00 46 4c 53 54 5f 53 45 51 2e 5a 54 32 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin

# part 1 - start of FLTS_SEQ in 7Byte->6Byte encoding
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part1.bin -t 1 ; hexdump -C part1.bin

# part 2
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part2.bin -t 1 ; hexdump -C part2.bin

# part 3
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part3.bin -t 1 ; hexdump -C part3.bin

# part 4
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part4.bin -t 1 ; hexdump -C part4.bin

# part 5
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part5.bin -t 1 ; hexdump -C part5.bin

# part 6
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part6.bin -t 1 ; hexdump -C part6.bin

# part 7
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part7.bin -t 1 ; hexdump -C part7.bin

# part 8
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part8.bin -t 1 ; hexdump -C part8.bin

# part 9
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part9.bin -t 1 ; hexdump -C part9.bin

# part 10
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part10.bin -t 1 ; hexdump -C part10.bin

# part 11
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part11.bin -t 1 ; hexdump -C part11.bin

# part 12
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part12.bin -t 1 ; hexdump -C part12.bin
[... 10 lines cut ...]

# part 15
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part15.bin -t 1 ; hexdump -C part15.bin

# part 16
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part16.bin -t 1 ; hexdump -C part16.bin

# part 17 - gives a shorter response... all 8502 bytes provided?
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part17.bin -t 1 ; hexdump -C part17.bin

# part 18 - gives a different response
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 22 14 2f 60 00 0c 00 04 00 00 00 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
amidi -p hw:1,0,0 -S "f0 52 00 6e 60 05 00 f7" -r part18.bin -t 1 ; hexdump -C part18.bin

amidi -p hw:1,0,0 -S "f0 52 00 6e 53 f7" -r temp.bin -t 1 ; hexdump -C temp.bin
```
Which gives the attached response(s)
[read_flts_seq.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3564119/read_flts_seq.txt)

mungewell, 2019-09-02:
more info
```
Part 1
--
00000000  f0 52 00 6e 60 04 22 01  00 00 04 00 3e 3e 3e 00  |.R.n`.".....>>>.|
                                            ^^ ^^ Start of 8->7 encoding
                                      ^^ ^^ Length (little endian, 7bit) = 512 bytes
00000250  2e 33 30 00 00 6f 6d 6e  58 03 f7                 |.30..omnX..|
                         ^^ ^^ ^^  ^^ ^^ Checksum? (8->7 encoded?)
                      ^^ last byte of encoded data

Part 15/16 identical
--
00000000  f0 52 00 6e 60 04 22 01  00 00 04 00 00 00 00 00  |.R.n`.".........|
                                      ^^ ^^ length = 512 bytes
00000250  00 00 00 00 00 07 15 56  6a 04 f7                 |.......Vj..|
                         ^^ ^^ ^^  ^^ ^^ Checksum?
0000025b = 603 bytes

decoded 603*7/8 = 527.625...
remove fixed parts gives us 603-12 = 591 *7/8 = 517.125
remove checksum 591 - 5 => 586 * 7/8 => 512.75?

Part 17
--
00000000  f0 52 00 6e 60 04 22 01  00 36 02 00 00 00 00 00  |.R.n`."..6......|
                                      ^^ ^^ Length = 2*128+54 = 310 bytes
00000160  00 00 00 00 00 00 00 00  00 00 00 00 00 00 63 75  |..............cu|
00000170  5c 6d 08 f7                                       |\m..|
00000174 = 372 bytes

372 - 12 -5 => 355 * 7/8 => 310.625
310 / 7 = 44.287 groups => 310 + 44 = 354 + 12 + 5 = 371 (too close to be wrong...)

16 * 512 + 310 = 8502 bytes total (magic!)
```

mungewell, 2019-09-04:
Can fully download FLST_SEQ via Midi, in Python with MIDO.

However the Midi packets appear to include a checksum, and I can't figure out what/how this is generated - does not appear to CRC32 or Alder32 which would be the natural choice. Reading does not require processing checksum, but I assume that it must be correct if sending an update FLST_SEQ.

Anyone got suggestions?
```
Nothing installed:
dummy.bin.hex:                  00000250  2e 33 30 00 00 1f 01 47  61 02 f7
                                                                      ^^ hi-bits
                                                         ^^ ^^ ^^  ^^ checksum +
1 effect installed, 1 bit different from 'nothing':
dummy_blackopt.bin.hex:         00000250  2e 33 30 00 00 5a 50 56  56 06 f7
dummy_comp.bin.hex:             00000250  2e 33 30 00 00 1c 6c 50  66 01 f7
dummy_greycomp.bin.hex:         00000250  2e 33 30 00 00 78 42 64  07 05 f7
dummy_lmt76.bin.hex:            00000250  2e 33 30 00 00 5b 1d 30  0b 03 f7
dummy_mutesw.bin.hex:           00000250  2e 33 30 00 00 04 25 2a  0a 05 f7
dummy_noisegate.bin.hex:        00000250  2e 33 30 00 00 22 10 7c  53 02 f7
dummy_optcomp.bin.hex:          00000250  2e 33 30 00 00 50 3b 68  20 04 f7
dummy_rackcomp.bin.hex:         00000250  2e 33 30 00 00 4a 75 46  76 07 f7
dummy_slowattack.bin.hex:       00000250  2e 33 30 00 00 3b 28 24  6b 02 f7
dummy_znr.bin.hex:              00000250  2e 33 30 00 00 08 69 62  17 0e f7

All zeros:
part16.bin.hex:                 00000250  00 00 00 00 00 07 15 56  6a 04 f7
```
[checksum.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3572453/checksum.zip)

mungewell, 2019-09-05:
Spent way too long staring at the checksum data, 'bent' into many different forms... seems Zoom's engineers are jokers and decided to invert the bits of the checksum (which is actually a CRC32 of the decoded data block - ie after 7bit -> 8bit conversion).
```
CRC32 decoded 0x952a57a5 10010101001010100101011110100101

Data from Midi b'5a50565606'
1011010 inverted                                  0100101
1010000 inverted                           0101111
1010110 inverted                    0101001
1010110 inverted             0101001
0110 inverted            1001
```
[checksum.py.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3577576/checksum.py.txt)

hardrubbish, 2019-09-05:
Wow. I'm completely illiterate when it comes to this sort of thing. It's amazing to watch you pick it apart and make sense of it all.

mungewell, 2019-09-06:
@hardrubbish Thanks, for me the 'Journey' is what makes these projects fun (despite being frustrating at times).

Late last night I managed complete an upload of a crafted FLTS_SEQ to my G1Four, although had error when uploading 'official' FLTS_SEQ (maybe my 8bit->7bit 'extra bits' code is wonky).

The real interesting observation is that I can add (a listing of) effect to FLTS_SEQ, but that does not mean it is operational on the pedal - it seems that the 'binary' can also be installed and removed (Official app seems to alway push the binary before updating FLTS_SEQ). This means that the pedal does not automatically fall back to the copy of the effect held in F/W flash.

So if I remove listing from FLTS and then re-add the effect will work (ie binary was already on pedal), but if I just add listing it won't.

Guess I need to implement the binary upload too ;-).

mungewell, 2019-09-07:
Fix the 'hi bits' problem. Attached is a snap-shot of what I currently have....

I implemented '-R' and '-S' commands to receive and send FLTS_SEQ to the pedal via Midi
```
C:\Users\simon\Documents\ZoomFW\zoom_zt2_20190907>"c:\Program Files\Python36\python.exe" zoom_zt2.py -h
Usage: zoom_zt2.py [options] FILENAME

Options:
  -h, --help            show this help message and exit
  -d, --dump            dump configuration to text
  -s, --summary         summarized configuration in human readable form
  -A ADD, --add=ADD     add an effect to group ADD
  -n NAME, --name=NAME  effect name (use with --add)
  -v VER, --ver=VER     effect version (use with --add)
  -i ID, --id=ID        effect id (use with --add)
  -u UNKNOWN, --unknown=UNKNOWN
                        effect unknown (use with --add)
  -D DELETE, --delete=DELETE
                        delete last effect in group DEL
  -w, --write           write config back to same file
  -R, --receive         Receive FLST_SEQ from attached device
  -S, --send            Send FLST_SEQ to attached device
```

For example, the following will 'delist' all effects on the pedal.
```
C:\>python.exe zoom_zt2.py -R -w current.ZT2
C:\>python.exe zoom_zt2.py -S empty.zt2
```

Note: The binary blobs for effects are still on the pedal and can be relisted. The next step will be to implement the uploading of the binary blobs.

[zoom_zt2_20190907.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3586858/zoom_zt2_20190907.zip)

mungewell, 2019-09-07:
Spoiler ;-)
[image]

mungewell, 2019-09-08:
It looks like effects can be uploaded to the stock F/W, without requiring it to be patched.

Download current state of FLST_SEQ ("-R -w"), then add new effect to it ("-A") and
uploaded it & the new binary blob ("-S" & "-I").
```
$ python3 zoom_zt2.py -R -w try_glamcomp.zt2
$ python3 zoom_zt2.py -A 1 -n "GLAMCOMP.ZD2" -v "1.10" -i 112 -u 128 -w try_glamcomp.zt2
Group 1:
    GLAMCOMP.ZD2
Add:GLAMCOMP.ZD2
$ python3 zoom_zt2.py -I "GLAMCOMP.ZD2" -S try_glamcomp.zt2
```
And it's installed... :-)

Zipfile also contains a "pre-built" exe, for those who don't have/want Python installed
```
C:\zoom_zt2_20190907b\build\exe.win-amd64-3.6>zoom_zt2.exe -h
Usage: zoom_zt2.exe [options] FILENAME

Options:
  -h, --help            show this help message and exit
  -d, --dump            dump configuration to text
  -s, --summary         summarized configuration in human readable form
  -A ADD, --add=ADD     add an effect to group ADD
  -n NAME, --name=NAME  effect name (use with --add)
  -v VER, --ver=VER     effect version (use with --add)
  -i ID, --id=ID        effect id (use with --add)
  -u UNKNOWN, --unknown=UNKNOWN
                        effect unknown (use with --add)
  -D DELETE, --delete=DELETE
                        delete last effect in group DEL
  -w, --write           write config back to same file
  -R, --receive         Receive FLST_SEQ from attached device
  -S, --send            Send FLST_SEQ to attached device
  -I INSTALL, --install=INSTALL
                        Install effect binary to attached device
  -U UNINSTALL, --uninstall=UNINSTALL
                        Remove effect binary to attached device
```
[zoom_zt2_20190907b.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3587182/zoom_zt2_20190907b.zip)

mungewell, 2019-09-08:
Note: Effect file name can not include a directory path, so 'ZD2' file should be in the local directory.

mungewell, 2019-09-08:
WARNING - After giving myself a minor heart-attack, I'll re-stress the comment above. Install ("-I") filenames should _JUST_ be the name, without any directory reference.

I had a script use "./COMP.ZD2" (etc) and fill all the internal memory, although these were not usable by the device. I was able to reclaim the space with "-U ./COMP.ZD2" and got back down to 22% used with all removed... guess there might be some other weird files stuck in there.

I'll see if there's a check I can implement in the code to prevent this.

mungewell, 2019-09-08:
Interesting snippet, using a star filename " * " will be interpreted as a wildcard (but not " A* "). Might be some fun to be had if there's a Midi command to get the next/subsequent match(es).
```
$ amidi -p hw:1,0,0 -S "f0 52 00 6e 60 25 00 00 2A 00 f7" -r prog.bin -t 1 ; hexdump -C prog.bin

41 bytes read
00000000  f0 52 00 6e 60 04 25 00  00 17 00 00 00 00 00 43  |.R.n`.%........C|
00000010  6c 61 70 2e 72 61 77 00  00 00 00 00 00 00 10 53  |lap.raw........S|
00000020  00 00 00 00 00 00 00 00  f7                       |.........|
00000029
```

mungewell, 2019-09-09:
@Barsik-Barbosik I am considering starting a project specifically for this python code, which would hopefully help track/improve it. What are your thoughts?

Barsik-Barbosik, 2019-09-10:
Wow! Amazing! I'm so happy that you found the solution for adding effects into the G1Four pedal!
It would be great and helpful, if you start and share your project with python code!

mungewell, 2019-09-17:
The 'other project' now has a GUI interface :-)
[image]

whm3223, 2020-03-02:
Hey sorry for having to ask this, but I've read this thread a number of times because I'm interested in putting bass amps on my G1X Four. Which file should I be editing and how do I find it?

mungewell, 2020-03-02:
> Hey sorry for having to ask this, but I've read this thread a number of times because I'm interested in putting bass amps on my G1X Four. Which file should I be editing and how do I find it?

@whm3223 
You will need to locate the appropriate '.ZD2' files for the amp/cabs that you want. This project would let you extract them from a B1Four F/W image (from Zoom's webpage).
```
Group 4 : AMP
    SVT_1U.ZD2 (ver= 1.00 ), group= 4 , id= 17 , installed= 1
    BMN1001U.ZD2 (ver= 1.00 ), group= 4 , id= 33 , installed= 1
    SWR4001U.ZD2 (ver= 1.00 ), group= 4 , id= 49 , installed= 1
    AG750_1U.ZD2 (ver= 1.00 ), group= 4 , id= 65 , installed= 1
    TE400_1U.ZD2 (ver= 1.00 ), group= 4 , id= 81 , installed= 1
    AC370_1U.ZD2 (ver= 1.00 ), group= 4 , id= 97 , installed= 1
    MINIMARK.ZD2 (ver= 1.20 ), group= 4 , id= 112 , installed= 1
    EBH3601U.ZD2 (ver= 1.00 ), group= 4 , id= 129 , installed= 1
    B15N_1U.ZD2 (ver= 1.00 ), group= 4 , id= 145 , installed= 1
Group 5 : CABINET
    SVT_8X10.ZD2 (ver= 1.20 ), group= 5 , id= 16 , installed= 1
    SV4X10TW.ZD2 (ver= 1.10 ), group= 5 , id= 24 , installed= 1
    FD_B4X12.ZD2 (ver= 1.20 ), group= 5 , id= 32 , installed= 1
    SWR_4X10.ZD2 (ver= 1.20 ), group= 5 , id= 48 , installed= 1
    AG4X10TW.ZD2 (ver= 1.20 ), group= 5 , id= 64 , installed= 1
    TE_4X10.ZD2 (ver= 1.20 ), group= 5 , id= 80 , installed= 1
    AC_1X18.ZD2 (ver= 1.20 ), group= 5 , id= 96 , installed= 1
    MKB2X8TW.ZD2 (ver= 1.20 ), group= 5 , id= 112 , installed= 1
    EB4X10TW.ZD2 (ver= 1.10 ), group= 5 , id= 128 , installed= 1
    AM1X15.ZD2 (ver= 1.10 ), group= 5 , id= 144 , installed= 1
```

Then try using the 'other' project to install them (there are binaries in the release section):
https://github.com/mungewell/zoom-zt2

clint-toy, 2020-03-03:
https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/issues/21#issuecomment-569381255

whm3223, 2020-03-03:
> [#21 (comment)](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/issues/21#issuecomment-569381255)

Thank you!!!