Barsik-Barbosik/Zoom-Firmware-Editor #17: More information on drum patterns?

mungewell, 2019-09-08:
The project mentions changing the drum patterns.... has any work been done on this?

Is there a 'file' within the F/W image (like a 'zdl' or something) which contains the patterns, and if so has any of it been decoded?

mungewell, 2019-09-18:
Zoom just announced the A1Four/A1FourX, which apart from different effects have a couple of other interesting things.
1). Drum patterns are different.
2). Looper can be placed either before or after effects chain.

Obviously no F/W available yet, but will good to look at these features when it is...

Tomlinsky, 2019-09-29:
They don't appear to be in the standard SMF format I would half have have expected,given their history of manufacturing rhythm machines. I could be wrong on this though, still investigating. I would expect patterns still to have delta trigger timings for hits of each pattern but without knowing the format this will,obviously, be tricky.

 This ties in with what I am trying to unravel. Knowing the actual structure of ZDL (in my case) or ZD2 files would open up a whole new world allowing new effects to be programmed. Much of them appear to be graphic data and crap spewed from the linker. Using simple effects, such as LineSel, would reveal much about the hardware froma coding point of view and I would love to be able to recreate the ZDL format.

 Willkeep nibbling away at it as and when I have time here. Great work by the way, your guys information is a great help.

Tomlinsky, 2019-09-29:
Oh, to answer your initial question... The drum patterns, samples, rhythm and looper(s) code are in the BIN folder of the firmware updaters self-contained resources. In the case of the G5n/G1Xon, at least,  it is filename "129".

PS: Pondering on my wish to code for the devices, all I actualy need is to know where the code segment is in a ZDL as I should then be able to use existing effects as a template. I assume a checksumor three isinvolved but will deal with that later. I should be abletoworkout how the hardware is mapped with that information. No definitve joy, sofar...

Tomlinsky, 2019-09-30:
Further information on drum patterns. In the resouce mentioned, areas appear to be dilineated by the ASCII string "YSX". Using SMF format, although logical and easier to mod, would have been inefficient as there is no need here for things such as individual note off, velocity, volume/etc information and complete overkill. All that is needed in a simple rhythm loop player like this are patterns of delta timings, which can be easily manipulated for tempo changes and a trigger number (1 of 24 in this case I believe). Each would only need to be 8 bits and I would expect at least an end marker for each pattern but that isn't a given as the player may be working from a list start addresses and lengths. There are a number of areas that look interesting. I'm only scanning in between other things here.

Also, my wish to get some code into the puppies is looking suspect at this point as I'm beginning to believe that code is decrypted by the boot ROM(s) upon transfer. In which case ZOOM did their job properly. * sigh *

Tomlinsky, 2019-09-30:
Okay,  I have found what I believe to be indexes to the rhythm patterns. But at this point I don't understand what file they are indexing or what the baseline for the index is. But it's a start. Perhaps.

mungewell, 2019-09-30:
Wow you seem to be 'storming ahead'! I attempted to find the '129' file, but my Java-Fu is weak. How can I get a copy of the file? Either by using the f/w tool, or downloading from the device.
For files like this it may be helpful to view as hex, as you say it only needs to contain 'events' for the drum hits... perhaps this are encoded into a binary blob.

Tomlinsky, 2019-09-30:
The installer executables are zip archives, just extract them. I actually have no use for this rhythm stuff personally but am intrigued at the moment. Need another cuppa, this actuallymay be simpler than I thought...

Tomlinsky, 2019-09-30:
Actually, being able to install a setup with less effects but more complex rhythms would be useful thinking about it....

Tomlinsky, 2019-09-30:
If you look at the section where the sample filenames are there is a string of 7 bytes directly before each pattern name each terminated by C0. There are possibly 8 - 10 bytes in the string but they all seem to be zeroes the first of which I suspect is a string terminator from the preceding filename. Some examples...

04 04 02 20 B7 0D C0  ; Guide
04 04 02 58 B2 0D C0  ; 8Beats1
04 04 02 88 87 0D C0  ; 8Beats2
04 04 02 18 87 0D C0  ; 8Beats3
04 04 02 58 85 0D C0  ; 16Beats1
04 04 02 E8 6A 0D C0 ; 16Beats2
04 04 02 48 6A 0D C0  ; 16Beats3
...
00 00 00 05 04 02 58 94 0D C0  ; CtWaltz1
00 00 00 03 04 02 78 A9 0D C0  ; CtWaltz2
03 04 02 B8 A2 0D C0  ; JzWaltz1
03 04 02 F0 AE 0D C0  ; JzWaltz2
03 04 02 98 A0 0D C0  ; Metro3
03 04 02 C8 B4 0D C0  ; Metro4
04 04 02 A8 B3 0D C0  ; Metro5
05 04 02 70 B2 0D C0  ; Metro

 Bytes 3-5 look like poiners to me as they are generally linear as you work through the list, C0 a terminator/identifier/padding with others control bytes of some kind. IF they are pointers.the only file big enough to contain data with a 24 bit offset is file "136" in the updater resources.

mungewell, 2019-09-30:
I think that the size and the contents show that the '129' file is very much more than just the drum patterns. There is a section which looks like it is assigning which sounds are used for which pattern...

```
00064D20: 00 03 04 02 30 02 18 C0  4A 7A 57 61 6C 74 7A 31  ....0...JzWaltz1
00064D30: 00 00 00 00 00 00 00 00  00 00 00 00 00 53 6E 72  .............Snr
00064D40: 41 6D 62 2E 72 61 77 00  00 53 6E 72 32 2E 72 61  Amb.raw..Snr2.ra
00064D50: 77 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  w...............
00064D60: 00 00 00 00 00 00 00 00  00 00 00 00 00 52 69 64  .............Rid
00064D70: 65 2E 72 61 77 00 00 00  00 00 00 00 00 00 00 00  e.raw...........
00064D80: 00 00 00 00 00 4B 69 6B  41 6D 62 2E 72 61 77 00  .....KikAmb.raw.
00064D90: 00 43 6C 69 63 6B 2E 72  61 77 00 00 00 03 04 02  .Click.raw......
00064DA0: 10 0F 18 C0 4A 7A 57 61  6C 74 7A 32 00 4B 69 6B  ....JzWaltz2.Kik
00064DB0: 32 2E 72 61 77 00 00 00  00 53 6E 72 41 6D 62 2E  2.raw....SnrAmb.
00064DC0: 72 61 77 00 00 53 6E 72  32 2E 72 61 77 00 00 00  raw..Snr2.raw...
00064DD0: 00 53 6E 72 34 2E 72 61  77 00 00 00 00 48 48 46  .Snr4.raw....HHF
00064DE0: 2E 72 61 77 00 00 00 00  00 52 69 64 65 2E 72 61  .raw.....Ride.ra
00064DF0: 77 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  w...............
00064E00: 00 4B 69 6B 41 6D 62 2E  72 61 77 00 00 43 6C 69  .KikAmb.raw..Cli
00064E10: 63 6B 2E 72 61 77 00 00  00 03 04 02 40 00 18 C0  ck.raw......@...
00064E20: 4D 65 74 72 6F 33 00 00  00 00 00 00 00 00 00 00  Metro3..........
00064E30: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064E40: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064E50: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064E60: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064E70: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064E80: 00 00 00 00 00 00 00 00  00 43 6C 69 63 6B 2E 72  .........Click.r
00064E90: 61 77 00 00 00 03 04 02  80 14 18 C0 4D 65 74 72  aw..........Metr
00064EA0: 6F 34 00 00 00 00 00 00  00 00 00 00 00 00 00 00  o4..............
```
I'd suspect that the numbers ahead of the name define the number of samples/etc, so these should be easily parsable.

mungewell, 2019-09-30:
So we are both typing at the same time... ;-)
BTW ZT2 files were using fixed length strings for file names.

Tomlinsky, 2019-09-30:
Oh it's a hell of a lot more than drum patterns , It contains the loopers, rhythm plaer, tuner parts of the deal as well.

mungewell, 2019-09-30:
Things I'd expect:
- Number of samples used (may be inferred), may also be normal/accented.
- Number of Beats in bar
- Number of Bars
- Maybe a pattern number (may be inferred)

Rather than a table of sample numbers, it may be a bit field of the drum sounds used in each beat.

One thing to question is whether the pattern info is before or after the pattern name... the last item seems to suggest after
```
00064F90: 98 10 18 C0 4D 65 74 72  6F 00 00 00 00 00 00 00  ....Metro.......
00064FA0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FB0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FC0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FD0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FE0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FF0: 00 00 00 00 00 00 00 00  00 00 00 00 00 43 6C 69  .............Cli
00065000: 63 6B 2E 72 61 77 00 00  00_04_04_01_D8_15_18_C0  ck.raw.......... <--- is this the data for Metro
00065010: 00 41 5F 69 6E 69 74 69  61 6C 69 7A 65 64 50 61  .A_initializedPa
00065020: 72 61 6D 73 3A 20 75 6E  69 6E 69 74 69 61 6C 69  rams: uninitiali
00065030: 7A 65 64 20 50 61 72 61  6D 73 20 73 74 72 75 63  zed Params struc
```

Tomlinsky, 2019-09-30:
It doesn't even need no of beats per bar or number of bars. It's only playing one  pass then retriggering. Delatas would do on their own. It could have that information but not really needed for such a simple player.

mungewell, 2019-09-30:
Looking at your list of "data; name", the data matches **previous** name.... waltz are in 3/4 time, and metro3/4/5 are 3/4/5 beats. All with 2 bars.

mungewell, 2019-09-30:
It may need to know time signature, as the looper can be set to bar/note lengths. Drum and looper tempo interact.

Tomlinsky, 2019-09-30:
It still only needs deltas. Not saying it doesn't have that info but that's how I'd do it in such a simple setup.

mungewell, 2019-09-30:
Do all the definitions have the same length (124 bytes), and same number of (potential) samples?

If we work on the assumption that data contains time sig and number of bars... what do the other 3 bytes mean? 0xC0 is likely a teminator, do the others imply offset (ie have nominal distance between them for 4/4 2 bars)?

```
000636E0: 46 75 6E 6B 32 00 00 00  00 4B 69 6B 31 2E 72 61  Funk2....Kik1.ra
000636F0: 77 00 00 00 00 53 6E 72  41 6D 62 2E 72 61 77 00  w....SnrAmb.raw.
00063700: 00 53 6E 72 33 2E 72 61  77 00 00 00 00 53 6E 72  .Snr3.raw....Snr
00063710: 32 2E 72 61 77 00 00 00  00 48 48 43 2E 72 61 77  2.raw....HHC.raw
00063720: 00 00 00 00 00 48 48 4F  2E 72 61 77 00 00 00 00  .....HHO.raw....
00063730: 00 48 48 43 41 6D 62 2E  72 61 77 00 00 4B 69 6B  .HHCAmb.raw..Kik
00063740: 41 6D 62 2E 72 61 77 00  00 43 6C 69 63 6B 2E 72  Amb.raw..Click.r
00063750: 61 77 00 00 00 04 04 02  B0 C7 17 C0              aw..........Funk
```
vs
```
00064F10:                          4D 65 74 72 6F 35 00 00  ....`...Metro5..
00064F20: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064F30: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064F40: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064F50: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064F60: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064F70: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064F80: 00 43 6C 69 63 6B 2E 72  61 77 00 00 00 05 04 02  .Click.raw......
00064F90: 98 10 18 C0                                       ....Metro.......

00064F90:             4D 65 74 72  6F 00 00 00 00 00 00 00  ....Metro.......
00064FA0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FB0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FC0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FD0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FE0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FF0: 00 00 00 00 00 00 00 00  00 00 00 00 00 43 6C 69  .............Cli
00065000: 63 6B 2E 72 61 77 00 00  00 04 04 01 D8 15 18 C0  ck.raw..........
```

mungewell, 2019-10-01:
Doesn't the pattern get displayed with dots, coloured in as pattern is played...?

Most are 4/4 time, what does "12/8 Grv" do (12/8 time)?

Tomlinsky, 2019-10-01:
Okay, updated installer. I think...

mungewell, 2019-10-01:
Another thing I note about the 'offsets'; yours mostly have '0x0D' as last byte and mine have '0x17' (B1 Four f/w).

Tomlinsky, 2019-10-01:
it's only bytes 3-4 possibly. May not be, but these are certainlynot the patterns... and they must be somewhere!

Tomlinsky, 2019-10-01:
Okay, I replaced  8Beats2 string with the one from 8Beats1 and they play exactly the same pattern.

Tomlinsky, 2019-10-01:
I reckon indexes, either...

04 04         02 20         B7 0D C0  ; Guide
04 04         02 58         B2 0D C0  ; 8Beats1
04 04         02 88         87 0D C0  ; 8Beats2
04 04         02 18         87 0D C0  ; 8Beats3
04 04         02 58         85 0D C0  ; 16Beats1
04 04         02 E8         6A 0D C0 ; 16Beats2
04 04         02 48         6A 0D C0  ; 16Beats3

or less likely (Although code is running in a TMS320C6745 and depending how the memory is mapped) ...

04 04         02 20 B7          0D C0  ; Guide
04 04         02 58 B2          0D C0  ; 8Beats1
04 04         02 88 87          0D C0  ; 8Beats2
04 04         02 18 87          0D C0  ; 8Beats3
04 04         02 58 85          0D C0  ; 16Beats1
04 04         02 E8 6A          0D C0 ; 16Beats2
04 04         02 48 6A          0D C0  ; 16Beats3

Tomlinsky, 2019-10-01:
Argh,GITHUB keepsstripping whitespace...

Tomlinsky, 2019-10-01:
Each pattern header is 124 bytes long containing the identified string of 7 bytes along sample filenames. However, a sample named in a kit does not necessarily mean it is actually used as evidenced by the first pattern header in the GFX1on...

GUIDE����Kik1.raw����SnrAmb.raw��Snr1.raw����������������HHC.raw�����������������HHCAmb.raw��KikAmb.raw��Click.raw���X²
À

mungewell, 2019-10-01:
I think your interpretation is correct (name, sample filenames), but I am not convinced about the other data. I think these bytes are (in order), time-sig1, time-sig2 and number of bars.... likely followed (as you suggested) by an offset in code/memory to the actual data defining the pattern.

Most have  '0x02', except for the 'Metro' pattern which has '0x01' - this one displays on device as "O o o o".
```
00064F90:             4D 65 74 72  6F 00 00 00 00 00 00 00  ....Metro.......
00064FA0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FB0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FC0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FD0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FE0: 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  ................
00064FF0: 00 00 00 00 00 00 00 00  00 00 00 00 00 43 6C 69  .............Cli
00065000: 63 6B 2E 72 61 77 00 00  00 04 04 01 D8 15 18 C0  ck.raw.......... = D8 15 18
```

The other oddity is '12/8 Grv' which displays as "O o o O o o O o o O o o"
```
00064940:                          31 32 2F 38 20 47 72 76  ........12/8 Grv
00064950: 00 4B 69 6B 31 2E 72 61  77 00 00 00 00 53 6E 72  .Kik1.raw....Snr
00064960: 41 6D 62 2E 72 61 77 00  00 53 6E 72 31 2E 72 61  Amb.raw..Snr1.ra
00064970: 77 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  w...............
00064980: 00 48 48 43 2E 72 61 77  00 00 00 00 00 00 00 00  .HHC.raw........
00064990: 00 00 00 00 00 00 00 00  00 48 48 43 41 6D 62 2E  .........HHCAmb.
000649A0: 72 61 77 00 00 4B 69 6B  41 6D 62 2E 72 61 77 00  raw..KikAmb.raw.
000649B0: 00 43 6C 69 63 6B 2E 72  61 77 00 00 00 0C 08 02  .Click.raw......
000649C0: E8 CA 17 C0                                       ....Waltz....Kik = E8 CA 17
```

mungewell, 2019-10-01:
The "offsets" are not in sequence, for the B1 Four F/W (apparently different to your dataset) they can be sorted as follows. The difference between them is not consistent, which might suggest a variable length data block.

```
simon@thevoid:~$ sort offset.txt 
 17 B8 D0  =  000645E0: D0 B8 17 C0 53 6B 61 00  00 00 00 00 00 4B 69 6B  ....Ska......Kik 
 17 BE D0  =  00064100: 61 77 00 00 00 04 04 02  D0 BE 17 C0 52 26 42 32  aw..........R&B2 
 17 C0 40  =  00064460: 63 6B 2E 72 61 77 00 00  00 04 04 02 40 C0 17 C0  ck.raw......@... 
 17 C0 F8  =  00063F10: 61 77 00 00 00 04 04 02  F8 C0 17 C0              aw..........ROCK 
 17 C2 60  =  00064370: 00 04 04 02 60 C2 17 C0  50 6F 70 00 00 00 00 00  ....`...Pop..... 
 17 C3 10  =  000634E0: 63 6B 2E 72 61 77 00 00  00 04 04 02 10 C3 17 C0  ck.raw.......... 
 17 C5 10  =  00063200: 00 04 04 02 10 C5 17 C0                           ........Rock1... 
 17 C5 B8  =  000643F0: B8 C5 17 C0 50 6F 70 52  6F 63 6B 00 00 4B 69 6B  ....PopRock..Kik 
 17 C6 60  =  00063E90: 63 6B 2E 72 61 77 00 00  00 04 04 02 60 C6 17 C0  ck.raw......`... 
 17 C7 08  =  000636D0: 63 6B 2E 72 61 77 00 00  00 04 04 02 08 C7 17 C0  ck.raw.......... 
 17 C7 B0  =  00063750: 61 77 00 00 00 04 04 02  B0 C7 17 C0              aw..........Funk 
 17 C8 58  =  00064750: 00 04 04 02 58 C8 17 C0  46 75 73 69 6F 6E 00 00  ....X...Fusion.. 
 17 C9 00  =  00063B30: 61 77 00 00 00 04 04 02  00 C9 17 C0 42 6C 75 65  aw..........Blue 
 17 C9 A8  =  00063180: 61 77 00 00 00 04 04 02  A8 C9 17 C0              aw..........16Be 
 17 CA 48  =  00063100: 63 6B 2E 72 61 77 00 00  00 04 04 02 48 CA 17 C0  ck.raw......H... 
 17 CA E8  =  000649C0: E8 CA 17 C0 57 61 6C 74  7A 00 00 00 00 4B 69 6B  ....Waltz....Kik 
 17 CB 80  =  00064940: 00 04 04 02 80 CB 17 C0  31 32 2F 38 20 47 72 76  ........12/8 Grv 
 17 CC 18  =  000637D0: 00 04 04 02 18 CC 17 C0                           ........JazzFunk 
 17 CC B0  =  000635E0: 00 04 04 02 B0 CC 17 C0  45 6D 6F 00 00 00 00 00  ........Emo..... 
 17 CD 48  =  00064200: 48 CD 17 C0 39 30 73 20  53 6F 75 6C 00 4B 69 6B  H...90s Soul.Kik 
 17 CD E0  =  000642F0: 61 77 00 00 00 04 04 02  E0 CD 17 C0 44 69 73 63  aw..........Disc 
 17 CE 70  =  00063850: 70 CE 17 C0 4D 6F 74 6F  77 6E 00 00 00 4B 69 6B  p...Motown...Kik 
 17 D0 20  =  000638C0: 63 6B 2E 72 61 77 00 00  00 04 04 02 20 D0 17 C0  ck.raw...... ... 
 17 D1 40  =  00063C30: 40 D1 17 C0 4A 61 7A 7A  31 00 00 00 00 00 00 00  @...Jazz1....... 
 17 D1 C8  =  000647D0: C8 D1 17 C0 53 77 69 6E  67 31 00 00 00 4B 69 6B  ....Swing1...Kik 
 17 D2 50  =  00064080: 63 6B 2E 72 61 77 00 00  00 04 04 02 50 D2 17 C0  ck.raw......P... 
 17 D7 58  =  00064180: 00 04 04 02 58 D7 17 C0  37 30 73 20 53 6F 75 6C  ....X...70s Soul 
 17 D7 D8  =  00063AB0: 63 6B 2E 72 61 77 00 00  00 04 04 02 D8 D7 17 C0  ck.raw.......... 
 17 D8 30  =  00063470: 30 DB 17 C0 4D 74 6C 43  6F 72 65 00 00 4B 69 6B  0...MtlCore..Kik 
 17 D8 58  =  00064270: 63 6B 2E 72 61 77 00 00  00 04 04 02 58 D8 17 C0  ck.raw......X... 
 17 DA B8  =  00064B30: 00 04 04 02 B8 DA 17 C0  42 72 65 61 6B 73 33 00  ........Breaks3. 
 17 DB A8  =  000644E0: 61 77 00 00 00 04 04 02  A8 DB 17 C0 45 75 72 6F  aw..........Euro 
 17 DC 20  =  00063370: 61 77 00 00 00 04 04 02  20 DC 17 C0 48 61 72 64  aw...... ...Hard 
 17 DC 98  =  00064010: 98 DC 17 C0                                       ....TomTomBt.Kik 
 17 DD F0  =  00063090: F0 DD 17 C0                                       ....16Beats1.Kik 
 17 DE 60  =  000633F0: 00 04 04 02 60 DE 17 C0  48 65 61 76 79 4D 74 6C  ....`...HeavyMtl 
 17 DE D0  =  000632F0: 63 6B 2E 72 61 77 00 00  00 04 04 02 D0 DE 17 C0  ck.raw.......... 
 17 DF 40  =  00063010: 00 04 04 02 40 DF 17 C0                           ....@...8Beats3. 
 17 DF B0  =  00062F90: 61 77 00 00 00 04 04 02  B0 DF 17 C0              aw..........8Bea 
 17 E0 20  =  00064AB0: 61 77 00 00 00 04 04 02  20 E0 17 C0 42 72 65 61  aw...... ...Brea 
 17 E0 90  =  00063280: 90 E0 17 C0 52 6F 63 6B  32 00 00 00 00 4B 69 6B  ....Rock2....Kik 
 17 E1 68  =  000639C0: 00 04 04 02 68 E1 17 C0  52 6F 63 6B 65 72 73 00  ....h...Rockers. 
 17 E1 D0  =  00064BB0: D0 E1 17 C0 35 2F 34 20  47 72 76 00 00 4B 69 6B  ....5/4 Grv..Kik 
 17 E2 38  =  000646D0: 61 77 00 00 00 04 04 02  38 E2 17 C0 53 68 75 66  aw......8...Shuf 
 17 F0 80  =  00063DA0: 00 04 04 02 80 F0 17 C0  42 6F 73 73 61 32 00 00  ........Bossa2.. 
 17 F0 E0  =  00063E20: E0 F0 17 C0 53 61 6D 62  61 31 00 00 00 4B 69 6B  ....Samba1...Kik 
 17 F2 00  =  000648C0: 61 77 00 00 00 04 04 02  00 F2 17 C0 31 32 2F 38  aw..........12/8 
 17 F3 10  =  00063660: 10 F3 17 C0 46 75 6E 6B  31 00 00 00 00 4B 69 6B  ....Funk1....Kik 
 17 F3 68  =  00063560: 61 77 00 00 00 04 04 02  68 F3 17 C0 46 61 73 74  aw......h...Fast 
 17 F5 78  =  00064560: 00 04 04 02 78 F5 17 C0  4E 65 77 57 61 76 65 00  ....x...NewWave. 
 17 F5 D0  =  00063A40: D0 F5 17 C0 32 6E 64 20  4C 69 6E 65 00 4B 69 6B  ....2nd Line.Kik 
 17 F6 28  =  00064650: 63 6B 2E 72 61 77 00 00  00 04 04 02 28 F6 17 C0  ck.raw......(... 
 17 F6 80  =  00064840: 63 6B 2E 72 61 77 00 00  00 04 04 02 80 F6 17 C0  ck.raw.......... 
 17 F6 D8  =  00064C20: 63 6B 2E 72 61 77 00 00  00 05 04 02 D8 F6 17 C0  ck.raw.......... 
 17 F8 F8  =  00063BB0: 00 04 04 02 F8 F8 17 C0  42 6C 75 65 73 32 00 00  ........Blues2.. 
 17 F9 40  =  00063940: 61 77 00 00 00 04 04 02  40 F9 17 C0 53 74 65 70  aw......@...Step 
[... cut ...]
 18 10 80  =  00062F10: 63 6B 2E 72 61 77 00 00  00 04 04 02 80 10 18 C0  ck.raw.......... 
 18 10 98  =  00064F90: 98 10 18 C0                                       ....Metro....... 
 18 12 60  =  00064F10: 00 04 04 02 60 12 18 C0                           ....`...Metro5.. 
 18 14 80  =  00064E90: 61 77 00 00 00 03 04 02  80 14 18 C0 4D 65 74 72  aw..........Metr 
 18 15 D8  =  00065000: 63 6B 2E 72 61 77 00 00  00 04 04 01 D8 15 18 C0  ck.raw.......... 
 18 15 E8  =  00062EA0: E8 15 18 C0                                       ....GUIDE....Kik
```

Tomlinsky, 2019-10-01:
The only reason I doubt the first 3 bytes are not bars/beats is that here funkier time sigs would be declared...

0C 08 02 88 6B 0D C0   ; Waltz - Possibly
05 04 02 58 94 0D C0  ; CtWaltz1 - ??
03 04 02 C8 B4 0D C0  ; Metro4 - ??
04 04 02 A8 B3 0D C0  ; Metro5 - ??

Again, only basing on how I would implement on a device like this, the idea of beats and bars in these devices is notional. It's triggering samples equal distances apart in time and resetting the start point at a predetermined time. All that is needed is a timer, some deltas that can be scaled for tempo and it will aways be in time with itself. We hear beats/bars but the code doesn't need to know the first thing about it as it can all be translated into "beats/bars" in a numerical form basically using division. If the patterns were linkable the that's a different kettle of doohdah.

But you could be right :)

Either way, that pattern data is somewhere.

Tomlinsky, 2019-10-01:
Actually, that can possibly be tested as well by putting the string for what is known to be 3/4 pattern into a 4/4 pattern header thinking about it.

Tomlinsky, 2019-10-01:
So, my Guide pattern nowplays a heavy metal waltz. I can replace the first three bytes with the original guide first 3 bytes and see what happens for a laugh...

mungewell, 2019-10-01:
It looks like you are using the data BEFORE the name, I think you should be look after.... ie. at the end of the block.
```
000649C0:             57 61 6C 74  7A 00 00 00 00 4B 69 6B  ....Waltz....Kik
000649D0: 31 2E 72 61 77 00 00 00  00 53 6E 72 41 6D 62 2E  1.raw....SnrAmb.
000649E0: 72 61 77 00 00 53 6E 72  32 2E 72 61 77 00 00 00  raw..Snr2.raw...
000649F0: 00 00 00 00 00 00 00 00  00 00 00 00 00 48 48 43  .............HHC
00064A00: 2E 72 61 77 00 00 00 00  00 00 00 00 00 00 00 00  .raw............
00064A10: 00 00 00 00 00 48 48 43  41 6D 62 2E 72 61 77 00  .....HHCAmb.raw.
00064A20: 00 4B 69 6B 41 6D 62 2E  72 61 77 00 00 43 6C 69  .KikAmb.raw..Cli
00064A30: 63 6B 2E 72 61 77 00 00  00 03 04 02 78 00 18 C0  ck.raw......x... = 78 00 18
                                      ^^ ^^ 3/4 time!
```

Tomlinsky, 2019-10-01:
Ah, the string is at the END of the header. There are strings before and after the headers here that have skewed things. Anyway, going to try changing the first 3 bytes to something from a known 4\4 string.

Tomlinsky, 2019-10-01:
Right, bars/beats and offsets it is. That fluff skewed all of the pattern associations.  What I'm not understanding, is why isn't a different pattern being played?

Tomlinsky, 2019-10-01:
Okay, the next section is the pattern data unless there is some voodoo going on here. Breaking it down into 32 byte chunks (with some yet undetermined  stuff at the end) an example of how I am currently interpreting the first few "patterns" run...

Dummy (There is a Dummy pattern in the G1Xon which also threw me earlier on)
2C D7 0D C0 00 09 00 00 B8 05 09 C0 B4 05 09 C0 B0 05 09 C0 C0 05 09 C0 C0 05 09 C0 A8 05 09 C0 
Guide (Pattern 1)
A2 05 09 C0 98 05 09 C0 90 05 09 C0 8A 05 09 C0 86 05 09 C0 C0 05 09 C0 C0 05 09 C0 C0 05 09 C0 
8Beats1 (Pattern 2)
C0 05 09 C0 C0 05 09 C0 82 05 09 C0 7A 05 09 C0 C0 05 09 C0 C0 05 09 C0 C0 05 09 C0 C0 05 09 C0 
8Beats2 (Pattern 3)
C0 05 09 C0 C0 05 09 C0 C0 05 09 C0 C0 05 09 C0 C0 05 09 C0 C0 05 09 C0 C0 05 09 C0 C0 05 09 C0 
8Beats3 (Pattern 4)
C0 05 09 C0 C0 05 09 C0 74 05 09 C0 6E 05 09 C0 68 05 09 C0 60 05 09 C0 54 05 09 C0 50 05 09 C0 
etc

Now, if I replace the pattern header string and pattern of "8Beats1" with the pattern header string and pattern of "Guide" (first pattern on G1Xon) when I update the firmware .... both patterns play the same!

Tomlinsky, 2019-10-01:
Upon further experimentation something not right there. Am back to the indexing idea on those extraneous , got a pattern header bytes. Will spend another evening on it, got a deadline coming up and always get drawn into distractions at these points for some weird psychological reason.

So...

38 42 65 61 74 73 31 00 00 ; Pattern Name
4B 69 6B 31 2E 72 61 77 00 00 00 00 ; Sample filename (Slot 1)
53 6E 72 41 6D 62 2E 72 61 77 00 00  ; Sample filename (Slot 2)
53 6E 72 31 2E 72 61 77 00 00 00 00 ; Sample filename (Slot 3)
00 00 00 00 00 00 00 00 00 00 00 00 ; Sample filename (Slot 4) 
48 48 43 2E 72 61 77 00 00 00 00 00 ; Sample filename (Slot 5) 
00 00 00 00 00 00 00 00 00 00 00 00 ; Sample filename (Slot 6) 
48 48 43 41 6D 62 2E 72 61 77 00 00 ; Sample filename (Slot 7) 
4B 69 6B 41 6D 62 2E 72 61 77 00 00 ; Sample filename (Slot 8) 
43 6C 69 63 6B 2E 72 61 77 00 00 00 ; Sample filename (Slot 9) 
04 04 ;Time Signature
02 ; Bars
88 87 0D
C0 ; Header terminator

mungewell, 2019-10-01:
Re
`0D ; ?? Device dependant constant`

I think that your earlier suggestion that this (and preceeding 2 bytes?) might be a pointer may be correct. The G1Four and B1Four both use 0x17 and 0x18 (ie different in the same file). The processor for these devices is know to be different and may have a different memory map.

Some of common deltas for these supposed pointers are:
56, 88, 112, 136, 152, 168

One thought on locating the pattern data, for simple patterns each bar is likely the same. So if it's not automatically repeated by the F/W then we should see some data repeated.

Oh, that procrastination monster ;-)

Tomlinsky, 2019-10-01:
Aye, a 24 bit address in a TMS320 memory map is perfectly normal. You would think the patterns should be optically identifialble indeed. They are definitely pointers, just swap any set of those numbers over here and the patterns change to different instrumentation if they are defined.

By the way, there are 70 pattern headers declared herebut not used by the firmware. One of the extra patterns is a dummy, the other called 12\8 Tri which is another shuffle pattern.

Tomlinsky, 2019-10-02:
The thing that deters me from belieiving they are 24 bit pointers is the howerver you interpret them the data range is too large for a bunch of timing data.

Tomlinsky, 2019-10-02:
A thought that I had earlier today was that they could be bit patterns for overlaying timing data. That would be more efficient on a simple player such as this. It would literally only need a dozen or so delta timing patterns to make enough variations for what are on offer here.

Tomlinsky, 2019-10-02:
Yep. I just replaced...

04 04 02 58 - [B2] - 0D C0  	; GUIDE

with...

04 04 02 88 - [87] - 0D C0  	; 8Beats1

and I get a kind of hybrid of the two.

Tomlinsky, 2019-10-02:
On a break here and wanted to leave some notes for posterity...

 The rhythm player appears to be a basic 9 channel tracker. Once the format of the data is worked out, which would need quite a bit of experimentation, this is actually an extremely flexible arrangement for building rhythm loops especially given access to variable bar lengths. 

 BUT, there's always one isn't there, for the idea of adding user rhythms to be of any real practical use in the G devices an external player/arranger would need to be impemented.user rhythms. This would need to recreate a firmware updater complete with new samples\pattern data. Even if it only used the built in patterns, coupled with changing samples\bar lengths\time sigs as a basis this could be pretty useful. There will be a limit to the delta time lengths and long patterns would need to be done within a "bar" as I imagine the player has no logic for linking them. It has no need, it isn't a MIDI player as that would be overkill and waste of resorces.

The following data is from a G1Xon\v1.21:

Pattern String format:
38 42 65 61 74 73 31 00 00 ; Pattern Name - (8Beats1)
4B 69 6B 31 2E 72 61 77 00 00 00 00 ; Sample filename Slot 1 - (Kik1.raw)
53 6E 72 41 6D 62 2E 72 61 77 00 00 ; Sample filename Slot 2 - (SnrAmb.raw)
53 6E 72 31 2E 72 61 77 00 00 00 00 ; Sample filename Slot 3 - (Snr1.raw)
00 00 00 00 00 00 00 00 00 00 00 00 ; Sample filename Slot 4 - (Empty)
48 48 43 2E 72 61 77 00 00 00 00 00 ; Sample filename Slot 5 - (HHC.raw)
00 00 00 00 00 00 00 00 00 00 00 00 ; Sample filename Slot 6 - (Empty)
48 48 43 41 6D 62 2E 72 61 77 00 00 ; Sample filename Slot 7 - (HHCAmb.raw)
4B 69 6B 41 6D 62 2E 72 61 77 00 00 ; Sample filename Slot 8 - (KikAmb.raw)
43 6C 69 63 6B 2E 72 61 77 00 00 00 ; Sample filename Slot 9 - (Click.raw)
04 04 ; Time Signature
02 ; Bars
88 87 ; Pattern Bitmap
0D ; ?? 
C0 ; Header terminator

Pattern Strings:

04 04 02 20 B7 0D C0  	; Dummy
04 04 02 58 B2 0D C0  	; GUIDE
04 04 02 88 87 0D C0  	; 8Beats1
04 04 02 18 87 0D C0  	; 8Beats2
04 04 02 58 85 0D C0  	; 16Beats3
04 04 02 E8 6A 0D C0 	; 16Beats1
04 04 02 48 6A 0D C0  	; 16Beats2
04 04 02 70 64 0D C0	; 16beats3
04 04 02 68 88 0D C0	; Rock1
04 04 02 38 86 0D C0	; Rock2
04 04 02 F8 83 0D C0	; Rock3
04 04 02 C8 85 0D C0	; HardRock
04 04 02 08 83 0D C0	; HeavyMtl
04 04 02 18 63 0D C0	; MtlCore
04 04 02 90 8F 0D C0	; Punk
04 04 02 50 6D 0D C0	; FastPunk
04 04 02 38 8F 0D C0	; Emo
04 04 02 68 66 0D C0	; Funk1
04 04 02 10 67 0D C0	; Funk2
04 04 02 B8 6C 0D C0	; FunkRock
04 04 02 F8 77 0D C0	; JazzFunk
04 04 02 18 79 0D C0	; Motown
04 04 02 B0 99 0D C0	; OneDrop
04 04 02 D8 88 0D C0	; Steppers
04 04 02 F0 90 0D C0	; Rockers
04 04 02 C0 7E 0D C0	; 2ndLine
04 04 02 60 68 0D C0	; Country
04 04 02 68 99 0D C0	; Blues1
04 04 02 A8 79 0D C0	; Blues2
04 04 02 E8 AA 0D C0	; Jazz1
04 04 02 60 A0 0D C0	; Jazz2
04 04 02 38 8B 0D C0	; Bossa1
04 04 02 98 8B 0D C0	; Bossa2
04 04 02 C0 65 0D C0	; Samba1
04 04 02 00 61 0D C0	; Samba2
04 04 02 88 9A 0D C0	; ROCKABLY
04 04 02 70 84 0D C0	; R'n'R
04 04 02 B8 7A 0D C0	; TomTomBt
04 04 02 68 5D 0D C0	; R&B1
04 04 02 40 7E 0D C0	; R&B2
04 04 02 E8 6D 0D C0	; 70s Soul
04 04 02 40 7F 0D C0	; 90s Soul
04 04 02 68 77 0D C0	; HipHop
[... 128 lines cut ...]
6C 9A 09 C0 6C 98 09 C0 88 C4 0A C0 80 C4 0A C0 A0 C4 0A C0 
78 C4 0A C0 70 C4 0A C0 60 C4 0A C0 68 C4 0A C0 A0 C4 0A C0 
60 C4 0A C0 CC C2 0A C0 C0 C2 0A C0 88 C2 0A C0 B4 C2 0A C0 
A8 C2 0A C0 8C C2 0A C0 98 C2 0A C0 88 C2 0A C0 8C C2 0A C0 
68 CC 0A C0 50 CC 0A C0 30 CB 0A C0 3C CC 0A C0 18 CC 0A C0 
FC CB 0A C0 DC CB 0A C0 C0 CB 0A C0 B4 CA 0A C0 88 CA 0A C0 
06 C9 0A C0 60 CA 0A C0 24 CA 0A C0 F4 C9 0A C0 C0 C9 0A C0 
22 4A 0A C0 16 4A 0A C0 1A 4A 0A C016 4A 0A C0 0C 4A 0A C0 
0C 4A 0A C0 12 4A 0A C0 38 23 09 C0 E4 21 09 C0 50 24 09 C0 
68 25 09 C0 68 25 09 C0 38 23 09 C0 E4 21 09 C0 64 FB 07 C0 
34 FB 07 C0 14 FB 07 C0 34 FB 07 C0 50 FB 07 C0 F4 FA 07 C0 
42 FB 07 C0 A8 09 09 C0 D4 09 09 C0 AC 09 09 C0 B0 09 09 C0 
B8 09 09 C0 C4 09 09 C0 CC 09 09 C0 94 B0 0A C0 88 B0 0A C0 
E6 AF 0A C0 60 B0 0A C0 50 B0 0A C0 44 B0 0A C0 A2 1B 0A C0 
F6 1B 0A C0 58 1B 0A C0 08 1C 0A C0 F0 1B 0A C0 E8 1B 0A C0 
26 48 08 C0 F8 47 08 C0 CC 47 08 C0 A8 47 08 C0 84 47 08 C0 
14 B1 09 C0 F8 B0 09 C0 78 B0 09 C0 E0 B0 09 C0 FA CF 0A C0 
E8 CF 0A C0 F4 CD 0A C0 38 CD 0A C0 B8 4C 0B C0 B0 4C 0B C0 
70 4C 0B C0 A8 4C 0B C0A0 63 81 11 0C 64 81 11 A0 63 81 11 0C 

64 81 11 50 B9 09 C0 
28 B9 09 C0 F8 B8 09 C0 CC B8 09 C0 
18 76 81 11 00 76 81 11 E0 75 81 11 C0 
75 81 11 24 AF 0A C0 
02 AF 0A C0 6C AF 0A C0 B4 AE 0A C0 D4 20 08 C0 E4 20 08 C0 40 21 08 C0 2C 21 08 C0

zeroeth, 2020-01-10:
@Tomlinsky Great work on the patterns! Will be fun to add pattern editing to the G1on!

re the zdl format, I've seen the same debugger strings in the firmware. Haven't dove in but I know the chip zoom uses is the TI tms320c6745ptp (tms320c6x family) at least on the ms50. And yeah it would be fun to pick apart a simple one. Best bet is to try and throw the zdl files at a disassembler that supports it!

mungewell, 2020-05-04:
Getting back to this whilst off work at the moment... wrote a little python script to decode that table of patterns/samples from the binary blobs of the F/W.
[decode_patterns.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/4572876/decode_patterns.zip)

It appears that the new v2.00 F/W contain _exactly_ the same data (G1 vs B1 vs G1X vs B1X), but the table is in a different location ('-O' for offset) to the v1.10 F/W.
`
C:\Users\simon\Documents\ZoomFW>python decode_patterns.py -O 407304 -d "B1 FOUR_v2.00\unzipped\.rsrc\1041\BIN\129" > "B1 FOUR_v2.00"\patterns.txt                       
C:\Users\simon\Documents\ZoomFW>python decode_patterns.py -O 407304 -d "B1X FOUR_v2.00\unzipped\.rsrc\1041\BIN\129" > "B1X FOUR_v2.00"\patterns.txt
C:\Users\simon\Documents\ZoomFW>python decode_patterns.py -O 407304 -d "G1X FOUR_v2.00\unzipped\.rsrc\1041\BIN\129" > "G1X FOUR_v2.00"\patterns.txt
C:\Users\simon\Documents\ZoomFW>python decode_patterns.py -O 407304 -d "G1 FOUR_v2.00\unzipped\.rsrc\1041\BIN\129" > "G1 FOUR_v2.00"\patterns.txt

C:\Users\simon\Documents\ZoomFW>python decode_patterns.py -O 405032 -d B1_FOUR_v1.10_Win_E\unzipped\.rsrc\1041\BIN\129 > B1_FOUR_v1.10_Win_E\patterns.txt
C:\Users\simon\Documents\ZoomFW>python decode_patterns.py -O 405032 -d G1_FOUR_v1.10_Win_E\unzipped\.rsrc\1041\BIN\129 > G1_FOUR_v1.10_Win_E\patterns.txt
`

Also v2.00 vs v1.10 has different values for the 'bit map' for each pattern.... I still suspect that this is a pointer to a location in memory where the actual pattern data is stored.

mungewell, 2020-05-04:
I did some looking at the 'offset' for the tables. Manually, and way too late into the night ;-)

As noted already the v2.00 F/W has different values for the offset compared to v1.10, however if you sort these offsets and then compare to calculate the number of bytes between them there is a very strong correlation between the F/Ws.

Mostly the drum pattern's data is the same 'length' (delta in 'offset'), although there are some slight changes for some. More complex patterns have a larger 'length'.
[image]

The annotated files are here:
[b1f_v200_patterns.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/4575485/b1f_v200_patterns.txt)
[b1f_v110_patterns.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/4575490/b1f_v110_patterns.txt)

mungewell, 2020-05-04:
I had made some copy+paste errors on the v1.10 report... now it's almost 100% correlation
[b1f_v110_patterns.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/4575894/b1f_v110_patterns.txt)

Offset is annotated "from -> to, length", "length" is almost identical. Will need to figure out where in the binaries these chunks of data are....
```
ListContainer: 							ListContainer: 
    ListContainer: 						    ListContainer: 
        Container: 						        Container: 
            name = u'Dummy' (total 5)				            name = u'Dummy' (total 5)
            offset = 1578848 (last in list)		      |	            offset = 1578472 (last in list)
        Container: 						        Container: 
            name = u'GUIDE' (total 5)				            name = u'GUIDE' (total 5)
            offset = 1577392 -> 1577416 24		      |	            offset = 1577088 -> 1577112 24
        Container: 						        Container: 
            name = u'8Beats1' (total 7)				            name = u'8Beats1' (total 7)
            offset = 1564992 -> 1565104 112		      |	            offset = 1564592 -> 1564704 112
        Container: 						        Container: 
            name = u'8Beats2' (total 7)				            name = u'8Beats2' (total 7)
            offset = 1564880 -> 1564992 112		      |	            offset = 1564480 -> 1564592 112
        Container: 						        Container: 
            name = u'8Beats3' (total 7)				            name = u'8Beats3' (total 7)
            offset = 1564432 -> 1564544 112		      |	            offset = 1564144 -> 1564256 112
        Container: 						        Container: 
            name = u'16Beats1' (total 8)			            name = u'16Beats1' (total 8)
            offset = 1559400 -> 1559560 160		      |	            offset = 1559112 -> 1559272 160
        Container: 						        Container: 
            name = u'16Beats2' (total 8)			            name = u'16Beats2' (total 8)
            offset = 1559240 -> 1559400 160		      |	            offset = 1558952 -> 1559112 160
        Container: 						        Container: 
            name = u'16Beats3' (total 8)			            name = u'16Beats3' (total 8)
            offset = 1558064 -> 1558232 168		      |	            offset = 1557776 -> 1557944 168
        Container: 						        Container: 
            name = u'Rock1' (total 5)				            name = u'Rock1' (total 5)
            offset = 1565216 -> 1565328 112		      |	            offset = 1564816 -> 1565032 216
        Container: 						        Container: 
            name = u'Rock2' (total 5)				            name = u'Rock2' (total 5)
            offset = 1564656 -> 1564880 224		      |	            offset = 1564368 -> 1564480 112
        Container: 						        Container: 
            name = u'Rock3' (total 5)				            name = u'Rock3' (total 5)
            offset = 1563968 -> 1564088 120		      |	            offset = 1563680 -> 1563800 120
        Container: 						        Container: 
            name = u'HardRock' (total 8)			            name = u'HardRock' (total 8)
            offset = 1564544 -> 1564656 112		      |	            offset = 1564256 -> 1564368 112
        Container: 						        Container: 
            name = u'HeavyMtl' (total 8)			            name = u'HeavyMtl' (total 8)
            offset = 1563728 -> 1563848 120		      |	            offset = 1563440 -> 1563560 120
        Container: 						        Container: 
            name = u'MtlCore' (total 7)				            name = u'MtlCore' (total 7)
            offset = 1557552 -> 1558064 512		      |	            offset = 1557264 -> 1557776 512
        Container: 						        Container: 
            name = u'Punk' (total 4)				            name = u'Punk' (total 4)
            offset = 1569936 -> 1570464 528		      |	            offset = 1569640 -> 1570168 528
        Container: 						        Container: 
            name = u'FastPunk' (total 8)			            name = u'FastPunk' (total 8)
            offset = 1560016 -> 1560168 152		      |	            offset = 1559728 -> 1559880 152
        Container: 						        Container: 
            name = u'Emo' (total 3)				            name = u'Emo' (total 3)
            offset = 1569848 -> 1569936 88		      |	            offset = 1569552 -> 1569640 88
        Container: 						        Container: 
            name = u'Funk1' (total 5)				            name = u'Funk1' (total 5)
            offset = 1558568 -> 1558736 168		      |	            offset = 1558280 -> 1558448 168
        Container: 						        Container: 
            name = u'Funk2' (total 5)				            name = u'Funk2' (total 5)
            offset = 1558736 -> 1558904 168		      |	            offset = 1558448 -> 1558616 168
        Container: 						        Container: 
            name = u'FunkRock' (total 8)			            name = u'FunkRock' (total 8)
            offset = 1559864 -> 1560016 152		      |	            offset = 1559576 -> 1559728 152
        Container: 						        Container: 
            name = u'JazzFunk' (total 8)			            name = u'JazzFunk' (total 8)
            offset = 1560464 -> 1560896 432		      |	            offset = 1560176 -> 1560608 432
[... 124 lines cut ...]
            name = u'CtWaltz1' (total 8)			            name = u'CtWaltz1' (total 8)
            offset = 1575144 -> 1575952 808		      |	            offset = 1574848 -> 1575656 808
        Container: 						        Container: 
            name = u'CtWaltz2' (total 8)			            name = u'CtWaltz2' (total 8)
[... cut ...]
            offset = 1577896 -> 1578488 592		      |	            offset = 1577568 -> 1578112 544
        Container: 						        Container: 
            name = u'Metro5' (total 6)				            name = u'Metro5' (total 6)
            offset = 1577416 -> 1577896 480		      |	            offset = 1577112 -> 1577568 456
        Container: 						        Container: 
            name = u'Metro' (total 5)				            name = u'Metro' (total 5)
            offset = 1578832 -> 1578848 16		      |	            offset = 1578456 -> 1578472 16

```

mungewell, 2020-05-04:
So, re-arranging the pattern names/offsets into sequential order we can see that the first difference in the "pattern table" between v2.00 and v1.10 is at "Rock1" (seems that pattern data may have been tweaked/swapped).

This means that there is about 9Kbytes (out of ~490K) which in theory are duplicated between files, but maybe at different locations.... thinking this might be a good(?) way to figure out which data block is for the patterns, Not sure about the best way to do such a comparison....

[image]

mungewell, 2020-05-07:
Ugh, spent way too much time staring into the abyss (aka hexdump of F/W). 

Still don't see where a table of notes/timing might be stored, but I am 100% convinced it must be there somewhere... 3 bytes (in the table we found) is not enough space to encode enough information.

But agree that my earlier comment about 8KBytes seems to be way too much space (considering FW is only 46KByte in total. Since the drum pattern table is in FW (129 file) it makes sense that the note/pattern data is also there somewhere.

I took samples of each of the drum patterns, to see what they do and maybe re-create them.
'Metro' proves that they have some adjustment of sample level, others show they have (at least) 16th note resolution, and lots of patterns are pretty complex.

Another hook which might help is that Metro3, Metro4 and Metro5 are very simple. Just 'click' repeated on the whole note at same/full volume.

mungewell, 2020-05-07:
Also '12/8 Tri' is in the FW table, but not listed in Manual or displayed by device... why?
[image]

Anyone found a FW file for the A1Four/A1FourX? This apparently has a different set of drum patterns.

mungewell, 2020-05-07:
Oh, so many questions. Looking at the 3rd data byte, what do these have in common?
```
$ python3 ../decode_patterns.py -d -O 407304 unzipped/.rsrc/1041/BIN/129  | grep -e name -e data3 | grep -B 1 '= 24' | grep name
            name = u'Dummy' (total 5)
            name = u'GUIDE' (total 5)
            name = u'Jazz1' (total 5)
            name = u'Jazz2' (total 5)
            name = u'Waltz' (total 5)
            name = u'CtWaltz1' (total 8)
            name = u'CtWaltz2' (total 8)
            name = u'JzWaltz1' (total 8)
            name = u'JzWaltz2' (total 8)
            name = u'Metro3' (total 6)
            name = u'Metro4' (total 6)
            name = u'Metro5' (total 6)
            name = u'Metro' (total 5)
```

Are they perhaps in a different area of memory?

Tomlinsky, 2020-05-07:
Eyup munge. Bit busy to dive back into this at the moment but my pitch is that the Zoom's have simple 1 'pattern' tracker type players in them each of which would only need to comprise of a pointer to a few bytes of trigger deltas/sample Nos/Sample Volumes followed by a terminator.. This would be a very compact solution especially as no handling of pitch information is required. Each pattern would effectively a 'bar' in tracker world with each consisting of only a few tens bytes even in a complex drum pattern.

mungewell, 2020-05-08:
Had some thoughts and scripted up printing out the 'pointers' I manually created before, which makes it easier to process every FW I can find.... one thought that hit me is that I am calculating/presuming the 'length' from the deltas between pointers.

What if the there is some other (random code/data/crap) between the sections of drum pattern data? This would mean that the 'length' would be overly large.

If we look at lengths across all the different FW, it's at least plausible that the pattern data would be (approx) same length on all pedals. So, overly large values would stand out.... for example.
```
16Beats1  160
16Beats2  160
16Beats3  168
2nd Line  128
5/4 Grv  856 <-----
5/4 Grv  544
5/4 Grv  1080 <------
5/4 Grv  1296 <------
70s Soul  152
70s Soul  2432 <------
```
This might mean that the sections of 'same data' might be harder to spot, or at least take more processing to find.
[patterns.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/4596788/patterns.zip)

Tomlinsky, 2020-05-08:
With the scalability of deltas being used for tempo and such a rigid 4/4 rhythm structure, for the most part, of triggers underpinning each simple pattern I would expect much repetition with sample trigger numbers changing the most.

mungewell, 2020-05-08:
might have made a break through, all 11 FW that I looked are think that some of the patterns are identical... this lead me to a section '129' memory which looks very interesting.

On the B1Four v1.10
```
00070630  00 81 00 71 03 f0
                            00 00  00 1d 00 5b 00 87 00 75  |...q.......[...u| <-- 144 bytes = "HipHop"
00070640  03 57 00 71 03 5b 00 75  03 3c 00 5d 00 26 00 77  |.W.q.[.u.<.].&.w|
00070650  06 57 00 71 03 5b 00 75  03 56 00 71 03 5b 00 75  |.W.q.[.u.V.q.[.u|
00070660  03 19 00 83 03 1c 00 5b  00 86 00 75 06 3c 00 5d  |.......[...u.<.]|
00070670  00 26 00 77 03 56 00 71  03 5b 00 75 03 58 00 72  |.&.w.V.q.[.u.X.r|
00070680  03 1c 00 5b 00 86 00 75  03 57 00 71 03 5b 00 75  |...[...u.W.q.[.u|
00070690  03 3c 00 5d 00 26 00 77  06 57 00 71 03 5b 00 75  |.<.].&.w.W.q.[.u|
000706a0  06 5b 00 75 03 19 00 83  03 1c 00 6c 00 86 00 46  |.[.u.......l...F|
000706b0  06 3c 00 5d 00 26 00 77  06 5b 00 75 03 18 00 58  |.<.].&.w.[.u...X|
000706c0  00 82 00 72 03 f0
```

On the G3nv2.20
```
000732d0  03 f0
                00 00 00 1d 00 5b  00 87 00 75 03 57 00 71  |.......[...u.W.q| <-- 144 bytes "HipHop"
000732e0  03 5b 00 75 03 3c 00 5d  00 26 00 77 06 57 00 71  |.[.u.<.].&.w.W.q|
000732f0  03 5b 00 75 03 56 00 71  03 5b 00 75 03 19 00 83  |.[.u.V.q.[.u....|
00073300  03 1c 00 5b 00 86 00 75  06 3c 00 5d 00 26 00 77  |...[...u.<.].&.w|
00073310  03 56 00 71 03 5b 00 75  03 58 00 72 03 1c 00 5b  |.V.q.[.u.X.r...[|
00073320  00 86 00 75 03 57 00 71  03 5b 00 75 03 3c 00 5d  |...u.W.q.[.u.<.]|
00073330  00 26 00 77 06 57 00 71  03 5b 00 75 06 5b 00 75  |.&.w.W.q.[.u.[.u|
00073340  03 19 00 83 03 1c 00 6c  00 86 00 46 06 3c 00 5d  |.......l...F.<.]|
00073350  00 26 00 77 06 5b 00 75  03 18 00 58 00 82 00 72  |.&.w.[.u...X...r|
00073360  03 f0
```

So anyone want to try decoding this???

Tomlinsky, 2020-05-08:
I'm on a different setup here now. Re <Also '12/8 Tri' is in the FW table, but not listed in Manual or displayed by device... why?> This was noted in the pattern structures previously. Can you give the dump for 8Beats1 instead of HipHop plz? I can refer to earlier data here . There looks to be 7 bytes of information before groups of 4 or 8 optically.

mungewell, 2020-05-09:
I added a '-D' option to my script to let user align the 'Int24ul' interpretation of the 3 bytes to location within the F/W '129' file. 
For the B1Four v1.10 FW this appears to work as:
`B1_FOUR_v1.10_Win_E>python ..\decode_patterns.py -O 405032 -D 454806 unzipped\.rsrc\1041\BIN\129`

[decode_patterns.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/4602764/decode_patterns.zip)
[locations.txt](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/4602761/locations.txt)

The '8Beats' pattern is
```
00071700: 06 5B 00 7B 06 F0
                            00 00  00 89 00 1E 00 78 00 5D  .[.{.........x.] "8Beats2"
00071710: 06 75 00 5A 06 29 00 78  00 3E 00 5D 06 87 00 1C  .u.Z.).x.>.]....
00071720: 00 75 00 5A 06 89 00 1E  00 78 00 5D 06 75 00 5A  .u.Z.....x.].u.Z
00071730: 06 29 00 78 00 3E 00 5D  06 75 00 5A 06 89 00 1E  .).x.>.].u.Z....
00071740: 00 78 00 5D 06 75 00 5A  06 29 00 78 00 3E 00 5D  .x.].u.Z.).x.>.]
00071750: 06 87 00 1C 00 75 00 5A  06 89 00 1E 00 78 00 5D  .....u.Z.....x.]
00071760: 06 75 00 5A 06 29 00 78  00 3E 00 5D 06 75 00 5A  .u.Z.).x.>.].u.Z
00071770: 06 F0 00 00 00 00
                            65 0C  00 1E 00 5D 00 89 00 78  ......e....]...x "8Beats1"
00071780: 06 5A 00 75 06 3E 00 5D  00 29 00 78 06 5A 00 75  .Z.u.>.].).x.Z.u
00071790: 06 1E 00 5D 00 89 00 78  06 1C 00 5A 00 87 00 75  ...]...x...Z...u
000717A0: 06 3E 00 5D 00 29 00 78  06 5A 00 75 06 1E 00 5D  .>.].).x.Z.u...]
000717B0: 00 89 00 78 06 5A 00 75  06 3E 00 5D 00 29 00 78  ...x.Z.u.>.].).x
000717C0: 06 5A 00 75 06 1E 00 5D  00 89 00 78 06 1C 00 5A  .Z.u...]...x...Z
000717D0: 00 87 00 75 06 3E 00 5D  00 29 00 78 06 5A 00 75  ...u.>.].).x.Z.u
000717E0: 06 F0 00 00 90 00
```

Note: the location for '8Beats1' may appear to be off, but it seems that Zoom likes to have a certain randomnes.... with other values the terminating '0xF0' can bleed into the beginning of a pattern.

mungewell, 2020-05-09:
maybe...
```
                                                     00 00  ....._.......... "Metro5"
00074860: 00 9F 0C 98 0C 98 0C 98  0C 98 0C 9F 0C 98 0C 98  ................
             ^^ ^^ 
             || ++- Delay to next note/sample
             |+---- Strength/Vol
             +----- Sample #
00074870: 0C 98 0C 98 0C F0
                         ^^ Terminator/Repeat
```

mungewell, 2020-05-11:
Made some adjustments to my script, it can now print out representations of the drum patterns,
[decode_patterns.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/4607082/decode_patterns.zip)
```
>python ..\decode_patterns.py -h
Usage: decode_patterns.py [options] FILENAME

Options:
  -h, --help            show this help message and exit
  -d, --dump            dump configuration to text
  -T TABLE, --table=TABLE
                        offset to pattern table within '129' file
  -D DRUMS, --drums=DRUMS
                        offset to drum data within '129' file
  -P, --pointer         print out the pointers to drum data as a sorted list
  -p PATTERN, --pattern=PATTERN
                        print drum data representation for a pattern
```

For example:
```
B1_FOUR_v1.10_Win_E>python ..\decode_patterns.py -T 405032 -D 454806 -p 3  unzipped\.rsrc\1041\BIN\129
Pattern 3 : 8Beats1 (0x00071776)
Bars 2 (4/4)
---
Kik1.raw   :#                       #     $                 #                       #     $                 :
SnrAmb.raw :            O                       O                       O                       O           :
Snr1.raw   :            #                       #                       #                       #           :
           :                                                                                                :
HHC.raw    :$     X     $     X     $     X     $     X     $     X     $     X     $     X     $     X     :
           :                                                                                                :
HHCAmb.raw :O     o     O     o     O     o     O     o     O     o     O     o     O     o     O     o     :
KikAmb.raw :O                       O     x                 O                       O     x                 :
Click.raw  :                                                                                                :

```

mungewell, 2020-05-13:
I was having some thoughts about what we know:
- Format of the table
- Format of the drum data
- Fact that drum data is interspersed with other data (presumably code)

The last bit means that it would be hard to replace sections without the risk overwriting important code, we could make all the pointers reference just one pattern and then overwrite the largest contiguous block.

I also noted that the patch data has an awful lot of unused space. If the area of memory could be identified, perhaps we could replace one of the patterns in the table with 'USER' pointing to the current's patch address...

It appears that the G1Four uses the same techniques/crc32 for the patch as it does for the FLTS_SEQ and file transfers. I have added code to my script to upload/download patches and un-encode them.
https://github.com/mungewell/zoom-zt2/blob/master/zoomzt2.py

peterjosvai, 2022-06-29:
guys, I'm not a programmer, unfortunately, but your conversation has been the most exciting read for me, ever :)
like reading about the Enigma :) 

I really admire your approach ---
Tomlinsky: reverse engineering in forward mode, i.e. inventing it and expecting it to be the same as Zoom's (incredible!)
Mungewell: using Python scripts to systematically reads through the code (amazing!)

It is also mind-blowing how you guys carry on with a thought kind of monologizing :) 

AND here lies something that could be a possible / viable breakthrough point

My impression was while reading your ship diaries :) that you guys felt to look of Zoom's engineers upon your screen...
and this gave you an extra level of excitement...
BUT... in my opinion, this extra excitement could be kind of blinding you.. cause you would expect at any moment someone come up and say: Tomlinsky, Mungewell, you're better than us!

which will never happen, as you know...

so, while feeling the heat of the reflectors -- their attention --  you were working in oblivion...

my suggestion:
please bear in mind that you might have been super close to finding the solution!!!!
much-much closer than you thought you were....

think of Electro-Harmonics: the are famous of using the cheapest parts (caps, ICs, transistors)...
ZOOM can't be so much different either!!

My guess is that it is your OWN creative minds which enlarged ZOOM's engineers' performance!!

Time after time, you cleared out the view, which is AMAZING!!!
and in those moments, ZOOM's product / composition appeared to be a brilliantly simple thing...

still there would be some not-fitting data, as I understood...
and the KEY to freely modifying the drum patterns, while having full control over the whole data structure would disappear, again and again...

my suggestion: DO NOT assume that ZOOM was counting with you!
I think they had to count with cheap Chinese copy-makers... but this is off their league... (I think)
with you, trying to unify G1ON and B1ON and edit the drum patterns???? I don't think so!!

so, my suggestion, forget any intentional obscurity... 
and assume something that they had to take care of instead...

and bear in mind that you might not see this something (which they had to take care of), because you're still not there....
and have not been testing it for months to find that some stupid misbehavior might show up here and then....
which they could fix by inserting some code in an ugly manner here and there :)

in any case: please, don't give it up yet!
take a go at it assuming that they don't watch you, and that they did not make any obscurity intentionally... (cause they thought that if someone had stolen it, they could easily  prove it before court that it was a copy of their relatively sophisticated work...)

also, please assume they they would use "common place" ZOOM solutions

I hope this comment does not hijack your beautiful thread... (which you should do from now on privately, I guess, so that you yourselves could believe that you don't assume their look upon you :) )

with respect!!! :)

P

[1 more reply]