Barsik-Barbosik/Zoom-Firmware-Editor #18: Unable to Extract V3 Effects from MS50G?

Jessafur, 2019-10-15:
Hi, I'm not sure if this is the right avenue to ask about this, but is there a specific way to get the additional effects added to V3 of the MS50G extracted? I cannot for the life of me figure it out. I have tried using the 3.00 and the 3.10 version of the firmware, but neither seem to have them. I also tried to load the Effect Manager, but that also would not work. Any guidance would be greatly appreciated.

Thanks!

hardrubbish, 2019-10-15:
So, you’ve opened the editor with it’s JAR file and pointed it to the firmware you want to extract the effects from?

Jessafur, 2019-10-15:
Hi!

Yes, I am using version 0.99.5 or the JAR and have successfully downloaded all of the effects from the MS60B, MS70CDR and all effects from firmware updates 1.00 and 2.00 of the MS50G. I have been trying to extract the additional effects using the below firmware (as well as the previous version 3.00). I have also tried extracting the version 3.00 effects using the MS50G Effect Manager, but the JAR does not recognize this file. Some of these effects I was able to copy from the MS70CDR, but I can't get any of the Drive or Amp Sim effects from version 3.00.

The specific effects that I cannot extract are as follows:

**DRIVES**

-CentaGold
-NYC Muff
-TS Drive
-BG Throttle
-OctFuzz
-BG Grid
-Red Crunch
-TB MK1.5
-Sweet Drive
-RC Boost
-Dynamic Drive

**AMPS**

-FD Vibro
-VX Combo
-BG Crunch
-Match 30
-Car Drive
-Tone City
-HW Stack
-Tangerine
-B-Breaker
-MS Crunch
-MS Drive
-BGN Drive
-BG Drive
-Revo-1

[E_MS-50G Effect Manager_Win_2.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3727640/E_MS-50G.Effect.Manager_Win_2.zip)

[MS-50G_v3.10_Win_E_2.zip](https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/files/3727625/MS-50G_v3.10_Win_E_2.zip)

hardrubbish, 2019-10-15:
Hmm, ok...
I did have some issues injecting effects from the 60B to the 70CDR so used effects from the B1ON which worked fine.
Maybe try sourcing the effects from the G/B1ON pedals.
If you haven’t already, maybe have a look at the issues thread I started about not being able to inject effects. Hopefully you can glean some info from it.

Jessafur, 2019-10-15:
I haven't had any issues actually injecting effects into the other pedals. It's simply that the effects that are only available on version 3.00 of the MS50G firmware do not show up in the editor. Some of the effects that I cannot extract from the 50G also do not come on the G/B1ON pedals, so that doesn't solve the problem either.

Is there any repository that has been posted here that has all of the effects pre-extracted?

ted210, 2019-10-15:
It's all in the readme file (last paragraph under getting started), including a link to a repository with all the effects pre-extracted: https://github.com/UnnoTed/zoom-ms50g/tree/master/efx_1_00

I extracted the latest version of MS-50G Effect Manager V1.3 (using the instructions in the readme), which includes updated versions of 2 of the effects (Silky Chorus and Dual Digital Delay).

Jessafur, 2019-10-15:
Hi, thank you! I somehow missed that part of the read me. Thanks for pointing me in the right direction!