Barsik-Barbosik/Zoom-Firmware-Editor #21: B1X Four not updating

neuman28, 2019-11-10:
I was able to download editor and make effect and amp changes to firmware update file. I saved the modified update file. When I double clicked to install the B1X Four went through all of the correct messages and the installer finished. I exited the installer and restarted the B1X Four.

No new effects were added and none that I removed were removed.

Current firmware is version 2.0

Any ideas what I can do? I really want to add a few additional effects.

Thanks.

vaesto, 2019-11-15:
There is another sister-project, which fix this issue. Search for "zoom-zt2" or mungewell (author).

neuman28, 2019-11-16:
Thanks. I think I understand how it should work but I have no knowledge of how to use the scripts or Python.

Sent via the Samsung Galaxy S®6 active, an AT&T 4G LTE smartphone

vaesto, 2019-11-16:
There are pre-compiled .exe inside the release, so you don't need to master the Python at all.

neuman28, 2019-11-18:
Thanks, didn't notice that. I'll give that a shot.

Sent via the Samsung Galaxy S®6 active, an AT&T 4G LTE smartphone

clint-toy, 2019-12-13:
Any success @neuman28 ?  

I tried the pre-compiled exe, got to add the effect I chose however the app gui freezes with error message on checksum.  The pedal works but the effect is not successfully added.

Some advice?

vaesto, 2019-12-13:
@clint-toy 
There are two variants based on different Python versions. The older (3.6) works, 3.7 - doesn't.

neuman28, 2019-12-14:
I have been busy with work and never tried. I will give it a shot tomorrow and let you know how it goes.

Sent via the Samsung Galaxy S®6 active, an AT&T 4G LTE smartphone

clint-toy, 2019-12-14:
looks like it was ver 3.6 i ran, however, same error.

neuman28, 2019-12-16:
Not a positive result here either. Headed out for holidays, try again in a few weeks.

Sent via the Samsung Galaxy S®6 active, an AT&T 4G LTE smartphone

clint-toy, 2019-12-28:
This is resolved by using the zoom-zt2 tool by mungewell.

First off is to extract the desired effects from compatible source firmware like from G1/x Four (I wish there is also for A1/x firmware to download, I got the MetalWRLD effect from G5 firmware :) ) using the Zoom-Firmware-Editor or other similar app.

Then, make sure you have enough space to inject new effects into the pedal by using the Zoom Guitar Lab app (or other tool with the same function) to remove/deactivate other effects.  The removed effects can be restored using the ZG Lab app should you want to revert.  (I removed some preamps, delays, reverbs, etc.)

Close other zoom apps/editors before opening the zoom-zt2 tool.  Then use zoom-zt2 to add/install the desired (extracted) effects into the pedal.  ( I got to install some of the guitar amps, cabinets, distortions, pedaldrive)

You can then open the Zoom Guitar Lab to manage the installed effects.  How I wish this device has more memory to store all sorts of effects :)