Barsik-Barbosik/Zoom-Firmware-Editor #25: not an issue, but a question

pezib, 2020-01-17:
hi Barsik-Barbosik,

now that Zoom G11 is announced at NAMM with integrated IRs module.
Do you think it will be a "classic" zoom module or a new architecture ?
In other words, if G11 uses a "standard kind of module", will it be adaptable to "old" zoom units ?

Thank you

Alexandre863, 2020-11-23:
Any solution ? I would love to have an IR Loader on my zoom Ms70cdr !

shooking, 2021-01-24:
I took a quick look at the updated for G11. Is it code signed now? Doesn't look like the other firmwares.

Which reminds me. Just like a Sony - I guess updating a pedal to latest firmware could eventually invalidate these tool chains.

I would hope Zoom can see these tools are ENHANCING their tools. I see a lot of musicians are experimenting with guitar pedals and synths - Keane have been doing this for years. And with some sysex / external hardware these pedals can be remote controlled from a synth. How cool is that?

asmerovski, 2022-12-12:
> 

Technically impossible, sorry to disappoint you.

asmerovski, 2022-12-12:
As for using pedals and amps from zoom G11, I managed to extract pedals and amps from G11 and put them to my zoom G3n. They are the same format and fully compatible. Not much difference in amount of pedals and amps (not including pedals with expression thingy). I remember I got Krampus amp, and couple more, they sound great. Couple of overdrives and fuzzes, but nothing much.