Barsik-Barbosik/Zoom-Firmware-Editor #26: Support GCE-3 'Guitar Circuit Emulator'

zeroeth, 2020-01-18:
I was poking around at the firmware here https://www.zoom.co.jp/products/production-recording/audio-interfaces/gce-3-guitar-lab-circuit-emulator

Seems to contain all the same structure as the large zooms, leads me to think its just a G#n series without any displays/buttons. Might be a cheap source of hacking once we figure out the ZD2 formats.

Tried to open the firmware exe with the editor, but didn't list anything.

shooking, 2021-08-06:
hi there
So I got one of these - loving it.
If you like then please follow my YouTube videos where I am reversing it.

https://www.youtube.com/watch?v=o0qB6FYXUSo

You can find more videos on the channel.

In the above I unbox, then add USBpCAP to look at what is going on.
Later I demonstrate using MidiOx to simulate a Pi4 (or other code) editing the GCE-3

I will edit another video over the weekend.
In this one I set the GCE-3 into B1XFour mode.
I create two patches, Pants 1 and Pants2, on my physical B1XFour at locations 51 and 52.
I dump the sysex via MidiOx.

Then I load up GCE-3 to demonstrate the content on the GCE-3's B1XFour - it is essentially factory.

Next I use MidiOx to load Pants 1 into the GCE-3.
And though and behold Patch 1 is now Pants 1.
I verify it is identical to my B1XFour patch.

Spurred on by this, I ask the MidiOx to interrogate the GCE-3's memory.
Can it find Glam Comp on 1-3? Sure.
Can it find Jaco on 1-7. SURE!

Now whether the system will recall this - will find out in the morning.
BUT you can guess what I will be doing - using the Pi 4 to run Mungewell's zoom-zt2 to determine if I can get the FLST etc etc.
And if so then I pretty much guarantee this GCE, in B1XFour emulation, respects the sysex.

Then I will see what happens if one emulated say a G5n.
Does this B1XFour patch corrupt? Or does the machine switch to a B1XFour. I wager it will corrupt.

In which case one would need to determine "hey ... what machine are you" and we know this question ...
<html>
f0 7e 00 06 01 f7
responding
f0 7e 00 06 02 52 6e 00 00 00 32 2e 30 30 f7
For a B1XFour 2.00

I think this little box is gonna be fun!

</body>
</html>

shooking, 2021-09-17:
I can pretty much control the GCE-3 with exception of looper, rhythms and external pedals (of which there is no input for a GCE-3 so no idea why bother to emulate X variants)
More on my youtube channel or ZoomPedalFun github

shooking, 2021-11-11:
Now I can handle the looper in terms of selecting the FX. Not sure how to use the footswitches necessary for start/stop. Zoom clearly are able to do it :-( But should be able to control the params.