Barsik-Barbosik/Zoom-Firmware-Editor #31: Effects not updating on MS-50G

w6ire, 2020-08-14:
I downloaded the editor and used it to create a modified version of the firmware for the MS-50G.  When connect my pedal to the PC using the firmware update mode and run the modified firmware file, it appears to work.  The screen on the pedal shows that the firmware update was successful, as does the message in the firmware software on the PC.  But no new effects are added to the pedal afterwards and none of the ones already installed are removed.

scuffedhelix, 2021-01-30:
I accidentally put my MultiStomp in this state too, though I was going out of my way to tinker with files. I'm now unable to change the available effects through firmware updates, but I am able to through [Zoom's MS-50G Effect Manager](https://zoomcorp.com/en/us/multi-effects/multistomp-pedals/ms-50g/ms-50g-support/). You'll be limited to MS-50G effects only.

yuziwen, 2021-10-01:
same issue....

ttshaw1, 2024-11-26:
I ran into this issue today. From what I gather from reddit, the common solution these days is [Zoom Effect Manager](https://zoomeffectmanager.com/en/). I'd prefer if this open-source solution worked, though; I don't suppose anyone's found a solution in the last 2.5 years?