Barsik-Barbosik/Zoom-Firmware-Editor #36: Patterns drums

cucubelo, 2020-10-24:
Hi how are you doing?
Is it possible to edit the drum patterns to have my own rhythm?
Thank you very much.-
I do not speak English

mungewell, 2020-12-01:
@cucubelo 
I don't think that anyone has done further work on actually building a FW with altered patterns. They are hard coded in the FW, for some of the G1Four (and friends) I worked out the offset(s) and the format enough to decode (and in theory encode) them.
https://github.com/mungewell/zoom-zt2/blob/master/decode_patterns.py

Running this script will output something like:
```
B1_FOUR_v1.10_Win_E>python ..\decode_patterns.py -T 405032 -D 454806 -p 3  unzipped\.rsrc\1041\BIN\129
Pattern 3 : 8Beats1 (0x00071776)
Bars 2 (4/4)
---
Kik1.raw   :#                       #     $                 #                       #     $                 :
SnrAmb.raw :            O                       O                       O                       O           :
Snr1.raw   :            #                       #                       #                       #           :
           :                                                                                                :
HHC.raw    :$     X     $     X     $     X     $     X     $     X     $     X     $     X     $     X     :
           :                                                                                                :
HHCAmb.raw :O     o     O     o     O     o     O     o     O     o     O     o     O     o     O     o     :
KikAmb.raw :O                       O     x                 O                       O     x                 :
Click.raw  :                                                                                                :
```

Unfortunately my pedal's screen broke, so I have not continued the work. Willing to help out if you are attempting this yourself.
Simon.

nopeg, 2024-08-09:
@mungewell 
Hello, I understand that it has been a long time since you last worked on a pattern decoding script, but I would really appreciate your advice on this topic because I would like to understand how everything works. I have little experience with hex decoding and whole working with memory stuff but I really liked the idea of changing hardcoded rhythms on my G1XFour. 
I used this project to remove all the effects (to free some space) and found decoded names for the patterns in 129 folder. Changed samples and names of the presets to test and it worked. I followed your issue #17 where you, Tomlinsky and other people figured out how to find offset to all the patterns and you did a great job! But I don't really understand how your script works with pointers and decodes all this stuff. 
Would appreciate any help!

mungewell, 2024-08-09:
@nopeg Yes, been quite a while, and the understanding/hacking on the G1Four and friends has been more active on another project. It may be better to open a ticket there with futher questions.

https://github.com/mungewell/zoom-zt2

The information on the drum patterns is still untested, I personally haven't tried overwriting the 'stock' patterns and don't know if anyone (yet) tried uploading a modified FW. Replacing the sounds should be simple, as that's just raw sound files on the pedals.

I wrote a script which can decode/extract the patterns:
```
simon@the-void:/media/simon/OS/Users/simon/Documents/ZoomFW/G1_FOUR_v2.00$ python3 ~/zoom-zt2-sdw-github/decode_patterns.py --help
usage: decode_patterns [-h] [-d] [-T TABLE] [-D DRUMS] [-P] [-p PATTERN] FILE

positional arguments:
  FILE                  File to process

options:
  -h, --help            show this help message and exit
  -d, --dump            dump configuration to text
  -T TABLE, --table TABLE
                        offset to pattern table within '129' file (G1Four V2.00 use 407304)
  -D DRUMS, --drums DRUMS
                        offset to drum data within '129' file (G1Four V2.00 use 457078)
  -P, --pointer         print out the pointers to drum data as a sorted list
  -p PATTERN, --pattern PATTERN
                        print drum data representation for a pattern
```

and can draw some ASCII art of a particular pattern:
```
simon@the-void:/media/simon/OS/Users/simon/Documents/ZoomFW/G1_FOUR_v2.00$ python3 ~/zoom-zt2-sdw-github/decode_patterns.py -T 407304 -D 457078 -p 0 unzipped/.rsrc/1041/BIN/129 
Pattern 0 : Metro (0x000756D6)
Bars 1 (4/4)
---
             :                                                :
             :                                                :
             :                                                :
             :                                                :
             :                                                :
             :                                                :
             :                                                :
             :                                                :
Click.raw    :$           $           $           $           :
```

The script uses 'construct' to `parse()` binary data into a Python dictionary, which can then be altered and converted back to binary with the `build()` function.

Further I don't know the location of the data within the G1X_Four FW image. I assume that it's the same content, so maybe a binary search could locate it...

mungewell, 2024-08-09:
Binary search...
```
simon@the-void:/media/simon/OS/Users/simon/Documents/ZoomFW/G1_FOUR_v2.00$ dd if=unzipped/.rsrc/1041/BIN/129 skip=457078 of=drums.bin bs=1 count=128
128+0 records in
128+0 records out
128 bytes copied, 0.000684528 s, 187 kB/s
simon@the-void:/media/simon/OS/Users/simon/Documents/ZoomFW/G1_FOUR_v2.00$ dd if=unzipped/.rsrc/1041/BIN/129 skip=407304 of=table.bin bs=1 count=128
128+0 records in
128+0 records out
128 bytes copied, 0.000663779 s, 193 kB/s
simon@the-void:/media/simon/OS/Users/simon/Documents/ZoomFW/G1_FOUR_v2.00$ ls -la *.bin
-rw-rw-r-- 1 simon simon 128 Aug  9 13:06 drums.bin
-rw-rw-r-- 1 simon simon 128 Aug  9 13:06 table.bin
simon@the-void:/media/simon/OS/Users/simon/Documents/ZoomFW/G1_FOUR_v2.00$ python3 ~/SearchBin-github/searchbin.py -f table.bin ../G1X_FOUR_v2.00/unzipped/.rsrc/1041/BIN/129 
Match at offset:         407304        63708 in  ../G1X_FOUR_v2.00/unzipped/.rsrc/1041/BIN/129
simon@the-void:/media/simon/OS/Users/simon/Documents/ZoomFW/G1_FOUR_v2.00$ python3 ~/SearchBin-github/searchbin.py -f drums.bin ../G1X_FOUR_v2.00/unzipped/.rsrc/1041/BIN/129 
Match at offset:         457078        6F976 in  ../G1X_FOUR_v2.00/unzipped/.rsrc/1041/BIN/129
```

So it looks like the tables are located at the same offset (of the '129' file) :-)
```
simon@the-void:/media/simon/OS/Users/simon/Documents/ZoomFW/G1_FOUR_v2.00$ python3 ~/zoom-zt2-sdw-github/decode_patterns.py -T 407304 -D 457078 -p 0 ../G1X_FOUR_v2.00/unzipped/.rsrc/1041/BIN/129
Pattern 0 : Metro (0x000756D6)
Bars 1 (4/4)
---
             :                                                :
             :                                                :
             :                                                :
             :                                                :
             :                                                :
             :                                                :
             :                                                :
             :                                                :
Click.raw    :$           $           $           $           :
```

nopeg, 2024-08-09:
>  the understanding/hacking on the G1Four and friends has been more active on another project
Yeah, I mixed up two repos. It is surprising that there are so many projects covering reflashing these processors and they are also somewhat connected :)