Barsik-Barbosik/Zoom-Firmware-Editor #38: Extract b1x and g1x

roy-lueders, 2020-12-08:
Hi,

if i extract the Firmware from b1x and g1x, but they are the same files?
you know why?

shooking, 2021-09-17:
I am making slow progress with the B1On here:
https://github.com/shooking/ZoomPedalFun/tree/main/B1ON/DerivedData

So we can extract the ZDLs from firmware (official or the Zoom Effect Manager).
We can work out the FXID/GID, version and name.
The effect manager can place them - I can only influence the FXID via sysex. Not worked out the GID yet. Else I have to create a patch and upload it.
But we can control FX parameters and a bit more.

I have a G5 on order - similar ZDL technology but works with Edit & Share so I will find out if it is possible to inject ZDLs into it and/or if the E&D sheds any light on additional Sysex
I would imagine the b1 and g1 are different firmware - but maybe instead they use some internal magic to not use some FX - except the Zoom Effect Manager allows me to add FX from similar class pedals.

clint-toy, 2022-02-18:
can we get the B6 amps/pre-amps/cabinets installed in B1x?