Barsik-Barbosik/Zoom-Firmware-Editor #41: Only 3 of the 4 possible effects are visible/usable?

AniMouseMusic, 2021-01-06:
Hi everyone,
My MS-60B works perfectly and I can add other effects than the stock version.
However, I cannot add more than 3. It simply does not show a 4th one in the row of effects. Am I missing something?
When I click on the left/right buttons, it does not go past 1 (obviously :-) but also not past 3??
I tried re-flashing it to the stock firmware (2.10), but that did not work.
I do not know if there is a way of fully resetting this pedal and if that would fix the problem.

It is working perfectly otherwise, so no real harm there. If it is not fixable, I would be happy just the way it is now.
But I still think it is a bit weird.

Any tips would be awesome!

shooking, 2021-01-24:
I dont know this pedal sorry. But as I look for more hints on mine (B1XFour g1xfour b1on) I happen across the other pedals.

On the first 2 above I found some sysex that cycles the FX slots. How? I was watching the USB traffic when using tonelib / GuitarLab

Assuming the stock pedal could have more than 3 do you have any presets with more than 3? And do tonelib or guitarlab work for your pedal?

If so u might try sniffing to see if you can find the strings.
Then try it on a modied 3 one. Maybe the slots are out by one. Some FX are two slots wide.

I have some videos on my YT of reverse engineering my Zoom pedal - i take through all the steps. Including some Python to process and harvest candidates.

Works with Windows 10. Should be adaptable to work with Linux. No idea about Mac. Couldnt get usbmon working as expected on Raspberry Pi4 - this is my preferred exploration platform now.

Sorry if I misunderstood your issue.

AniMouseMusic, 2021-01-24:
Hi there, thanks for your reply.
Just a few days ago, I managed to fix the problem by pressing and holding the left "next" button. Suddenly, a wild 4th effect appeared! Pressing that button was very effective :-)
Ok, jokes aside, it seemed to be hidden or something, but now I can use it. 
I think it might have something to do with the preset, like you mentioned. Maybe the one I started with only had 3 effects active.
The MS-60B is the bass version of the ms-50g and probably a close relation to the B1XFour. (Not 100% sure, but I think the effects would sound similar)
Anyways, long story short, thanks for the help and it works now :-)

Feel free to share your YouTube channel though, as I am curious to see your projects!
Greetings Maurice

shooking, 2021-01-24:
Ok well you did ask for it (be aware I am a bit Asperger's so yes/no is tricky for me).
https://www.youtube.com/watch?v=B5W7c5dEVmI

This video I take step by step with Windows on how to use USBPcap to capture the packets. I assume you can load Wireshark and copy USBPCap - the description in the video covers the steps but if you want to follow along and something is missing do let me know.

Capturing at the USB level means you see everything passing to/from pedal (but caution since the pedal deliberately filters -- something I dont understand :-)).
This is when you have to turn on EditorMode to make it spill the beans.

The raw midi is USB midi. This is NOT the same as cable midi. I hack the protocol - miss every 4th byte. This is technically wrong but close enough for initial looks.

Once you are capturing as Admin user, start your favourite external software and watch what it does.

Typically they send "Identify" until they get some response.
This allow your device ID to be read.
Then they send an EditorOn and typically PCMode on.
Then they start at patch 0 ... get contents.
patch 1 .. get contents 
....

then on my Zoom B1X it sometimes sends "reset iterator, step thru FX until nothing comes back". I not quite worked this out for B1On yet.

Barsik hacks the firmware to inject FX on block boundaries (at least that seems to be what is happening).
Mungewell found you can also read content of pedal via Midi.

What I am doing is reversing the pedal and their code - via collaboration - as I want to use Raspberry Pi 4 to control the pedals and a touch screen. You can see later videos on this subject on my channel.

I have to use Windows if I want to load firmware / use GuitarLab or use ToneLib.

But I can now also investigate quite a lot of the pedal directly on the Pi 4. And this is a cheap (and warm because this is in the house, not in the garage) was to play along.
A Pi 4 4GB would be minimum I recommend but I got 8 GB cos I was bored during lockdown - wanted to port London Undergound train driving sims for my amusement.

Take what I am doing right now:

Get Pi 4 to watch the incoming traffic.
Set up a blank patch on B1On.
Press front panel buttons to cycle thru the FX.
Keep going until you get back to Bypass.

Now I have a file of "patches" and can see it is simply sending a complete patch where 1st slot is replaced with an "on firmware" image of an FX.

It should be possible to get a ToC from the firmware, from the pedal or even with this crappy method.
From that, and reading manual, the folders, groups and FX ID hopefully can be reversed.
Then one can send sysex to set a given FX at a given slot.
How to put it on a patch?
Ask B1On to move to the patch ... then do above.

There seems to be very similar Sysex functions across the pedals. They get a bit more sophisticated over time, the device ID changes, and the command too. But in principal the devices behave similarly.

If you want to play along then great.

AniMouseMusic, 2021-01-25:
Hi there!
First off, no worries about the detailed explanation :-)
The programming part is a bit out of my scope. What I have done with my pedal is switching patches, and I tried this software and another program.
I ended up adding the effects that I wanted, so my pedal is good to go.
It is maxed out with all possible effects (Some extra reverbs/vibrato/lo-fi effects from the MS-50G), but no more room for some effects from other pedals :-) 

Only thing left to do is to make music with it! It sounds great for a pedal this size.
What instruments do you play (if I may ask?)

shooking, 2021-01-25:
Well see the memory of these things gets maxed out quickly. I totally agree
with your approach - set and leave alone. My plan to use the touch panel is
to also allow loading patches from disk. Or modify some parameters real
time - useful for synths and general control.

I used to play keyboards (spell checker replaced it with kebabs ... AI is
smart these days)

Now I play bass at a church and occasionally electric guitar when they
aren't looking.
I have some edrums  - not so good at that but nice stress release.

I like building kit to control synths / FX. Hence why I am lurking here.

AniMouseMusic, 2021-04-10:
Hey man, sorry for the delay.
Sounds all good :-) (Pun intended)
Currently I am looking into modding the pedal to add an external pedal
up/down type of thing.
Weird part is that you can only switch between effects within a patch, not
the patch itself. (Which in my opinion would make more sense?)
However, switching the effects in a patch can be helpful to in some
situations I guess. And, it is not a dirty mod if you do it right.

So, that is my current position with the pedal. Still having it packed full
of effects I will never use, haha. I think I will end up using around 10 or
20 MAX!
But I can imagine that for keyboards/synths, this pedal can be ofmore value
with a ton of effects.
For guitar/bass, only a few are usable in my music, mostly the
verb/vibrato/retro style sounds and for bass, some compressors and/or dirt.
That is basically it.

Anyways, I just thought to update you on that part, haha.
Btw, I noticed you bought a Harley Benton guitar too! What a coincidence. I
just bought their Telecaster model for dirt cheap.
Probably one of the best guitars I have in my rig! They are a surprisingly
good company, especially for the value.
Keep up the good work!
Btw, I just commented on your channel with my YouTube for Musicians
channel. So, that is me, haha.

Greetings Maurice

shooking, 2021-04-10:
Thanks.
I created new github with wiki and code. Aim is to give reference and deeper dive.
I not been looking specifically at the MS60 but I give the links to super work in that area.

https://github.com/shooking/ZoomPedalFun

Today I was extending Mungewell's decoding of the ASCII dump for B1XFour but using C. For these modern pedals Mungewell has the atomic operations to get a lot out of them. Need to make some time with the Ctrlr panel layout.