Barsik-Barbosik/Zoom-Firmware-Editor #42: A1 Four effects

b4l00ney, 2021-01-23:
So I found this file: https://gofile.io/d/jfxs8Y

Looks like it's a firmware dump of the A1 Four.

I don't have a compatible pedal, but can anyone test if the effects from there work?

scuffedhelix, 2021-01-30:
Where did you find this file? I'd love to figure out how to dump my MultiStomp firmware.

I had .ZD2s ripped from an A1X Four via @mungewell's [zt2 tool](https://github.com/mungewell/zoom-zt2). They worked fine in my G1X Four, except for the acoustic modelling ones which since then have [received support](https://github.com/mungewell/zoom-zt2/commit/d74429ceaa33360c9832eb3398a7f27a3f7480a9).

mungewell, 2021-05-08:
You could try this...
https://github.com/fuzboxz/zdownload

adamkup, 2021-08-12:
> Where did you find this file? I'd love to figure out how to dump my MultiStomp firmware.
>
> I had .ZD2s ripped from an A1X Four via @mungewell's [zt2 tool](https://github.com/mungewell/zoom-zt2). They [...]

Hi  scuffedhelix, 

Could you share the ZD2 files that you got of the A1?

sfinotti, 2022-05-28:
> Where did you find this file? I'd love to figure out how to dump my MultiStomp firmware.
>
> I had .ZD2s ripped from an A1X Four via @mungewell's [zt2 tool](https://github.com/mungewell/zoom-zt2). They [...]

Hi scuffedhelix,
Could you share the ZD2 files that you got of the A1?

mungewell, 2022-05-28:
@adamkup You might download this file...
https://www.zoom.co.jp/archive/GUITAR_LAB/AllZDL6.lst

And then look at the top...
```
{  
   "zdl_url":"http://www.zoom.co.jp/archive/GUITAR_LAB/ZDLF3/",
   "models":[  
```

And then midway in the file....
```
      {  
         "modelId":"0x6E001200",
         "modelName":"A1X FOUR",
         "jpegImage":"http://www.zoom.co.jp/archive/GUITAR_LAB/model_image3/A1X_FOUR.png",
         "editor":true,
         "patch":true,
         "effects":[  
            {  
               "type":"0x8000021",
               "fileName":"ANDLY_3S.zip",
               "version":"1.10"
            },
```

And put the `zdl_url` and `fileName` together for each effect that you want.

sfinotti, 2022-05-30:
> 

> 

Hi @mungewell 
Thanks for the fast answer !!!!
So, if I want to add the A1X FOUR effect "ANDLY_3S" I'll have to edit the file "AllZDL6.lst" to add the "complete path" for the zip file, the following way ??

      {  
         "modelId":"0x6E001200",
         "modelName":"A1X FOUR",
         "jpegImage":"http://www.zoom.co.jp/archive/GUITAR_LAB/model_image3/A1X_FOUR.png",
         "editor":true,
         "patch":true,
         "effects":[  
            {  
               "type":"0x8000021",
               "fileName":"http://www.zoom.co.jp/archive/GUITAR_LAB/ZDLF3/ANDLY_3S.zip",
               "version":"1.10"
            },

I'm planning on buying a pedal to use with acoustic guitar, but here in Brazil the A1FOUR is way too expensive (2x or even more than the G1FOUR price). So, if it's possible to load the A1FOUR effects into a G1FOUR I'll probaply go that way....

Once again thak you !!!

mungewell, 2022-05-30:
@sfinotti 
Same here in Canada, the A1Four is quiet a bit more, but it includes the XLR to 1/4inch (active) adapter.

You don't specifically need to edit the file, just use the information in it...

If you click on "combined" link in the above entry, you will end up with a small zip file on your drive which contains that specific effect. This project will allow you to upload the unzipped effect ('.ZD2') to a G1Four (or the like).

sfinotti, 2022-05-30:
@mungewell mung
Ok.... so I just have to download the files, unzip it and use your tool (ZoomZT2) to load it to the Pedal.
BTW I downloaded the files from project ZoomPedalFun and it apperas that all the effects (or at least a lot of them) from A1FOUR is there.... 

It seems that those pedals (A1FOUR, B1FOUR and G1FOUR) are almost identical, except for the mic input on A1FOUR.
I could not find hardware specs nor service manuals / schematics to confirm that.... so it's just a guess...

mungewell, 2022-05-30:
Yes, you got it. 

I too believe that the devices are identical apart from the plastic colour and labelling, the "X" versions have additional components on the circuit board for the expression pedal. The MIC input on the A1Four is actually an external adapter/cable.

I have a listing of all known effects, and those specifically marketed with the A1[X]Four...
https://github.com/mungewell/zoom-zt2/blob/master/zoom_fx/master.txt
https://github.com/mungewell/zoom-zt2/blob/master/zoom_fx/A1X%20FOUR/unzipped/list_sorted.txt

There is another file within the unit (".BDL") which I believe is a EQ on the input, definitely different between the G1Four and and the B1Four. May be different on the A1Four. Unlikely to be a big difference though.

sfinotti, 2022-05-31:
> There is another file within the unit (".BDL") which I believe is a EQ on the input, definitely different between the G1Four and and the B1Four. May be different on the A1Four. Unlikely to be a big difference though.

I saw an yt video mentioning that input EQ. This one is independent and wont mess with the specific settings.... Seems interesting !!!

shooking, 2022-06-08:
hi @sfinotti 
I already unpacked them from a Zoom GCE-3 here if you want to take a look?

sfinotti, 2022-06-09:
Hi @shooking 
Thank you for the reply !!
Yes I had already downloaded them from your project (great project, by the way) and mentioned that 4 posts above (https://github.com/Barsik-Barbosik/Zoom-Firmware-Editor/issues/42#issuecomment-1141476114).
You just mentioned a point about unpacking the effects from GCE-3. So GCE-3 uses the same effects as A1FOUR ???

shooking, 2022-08-06:
Hi sorry - I must have missed your question. If you turn a GCE-3 into an A1XFour then sure - it is an A1XFour.
But Mungewell's already shown how tooad FX from one compatible pedal to another. The GCE-3 just left you grab all the FX. You can then move them to a physical pedal or even to the emulated version. Guitar Lab 7 started to show the injected FX. Earlier versions would show missing FX.

Milan160, 2022-12-01:
Hi, is there any way to add effects from G1X FOUR to A1X FOUR? it doesn't exist anywhere, I couldn't find the firmware.exe for ZOOM A1X Four that I need to be able to transfer effects via Zoom Firmware Editor to the A1X Four pedal. Can someone help me?

nomadbyte, 2022-12-30:
> @mungewell: There is another file within the unit (".BDL") which I believe is a EQ on the input, definitely different between the G1Four and and the B1Four. May be different on the A1Four. Unlikely to be a big difference though.

Anyone tried to download the BDL file directly off the A1/A1X FOUR? Not sure what's the exact name, but I'd guess `A_OUT_EQ.BDL`.

TinyLightBold, 2023-06-17:
Would it be possible to load the effects from A1x into the G5n?
The filesystem match ofcourse. And how about the patches?

shooking, 2023-06-17:
You would have to load the FX of the A1X onto the G5n then it would be easier to not the fx, position in chain and values and set these via the G5n.

nomadbyte, 2023-08-23:
@TinyLightBold : A1 FOUR ZD2 modules should be compatible with G5n. However the patches are not directly compatible. Technically the patches are convertible but that needs a separate processing.