Barsik-Barbosik/Zoom-Firmware-Editor #46: ZDL file format

shooking, 2021-08-05:
Hi @Barsik-Barbosik 

I've been working with @mungewell looking at ZDL files - specifically for the B1ON
Thanks for posting hints on the file format of the updater.

Looking at what both of you have found it seems some FLST_SEQ.ZDT determines available FX in a firmware (could this mean FX list?).
And it seems one should order by Group ID (GID) and then alphabetically FXName.

So an updated file has some preamble to make Zoom recognise it.
Often a FLST_SQL.ZDT (or variant)
A list of ZD2/ZDLs in group, then name order
On the B1XFour etc Mungewell found there is an ID that can be divided into FXID and GID.
I dont find the same IDs from the B1On firmware

QUESTION

Do we know if the FXID of an FX is encoded directly in the file itself?
That is can a ZD2 be taken in isolation and from that the FX/GID derived?
Is this similar for a ZDL

And ultimately what is of interest in a ZDL?

BACKGROUND

I managed to use another program to insert ZDLs into Zoom B1On.
I had made quite a bit of progress documenting the Sysex for these pedals, and indirectly working out the FXID/GID.

Mungewell confirms there is some sort of FLST_SEQ in the 1.30 firmware.
But it isnt so clear that the IDs are there.

Now, when I revisited the topic recently, but with the CDR injected ZDLs, I could see the FX/GID has changed - seems to have.
And yet the filenames are same.

Last night I found how to extract the ZDLs from a firmware.
Today I plan to compare them.
I was hoping to find something that correlates with the FX/GID I get from the pedal.

NOTE: the first 4 bytes of the unpacked FX seem to be the FX/GID but you have to 0 out the parameters to see it.
I have sysex to control the 1st two bytes but not the next 2.
So even if I know the FX/GID I can only set FXID and that least to "missing FX".

I am reasonably sure most of the control programs (1-Controller, Tonelib) are sending full sysex to update FX - not interrogating
And hence they can never be in sync like the zoom-zt2 can.

I would love to find a way to change that on the B1On.

mungewell, 2021-08-05:
Adding a little more background, this is where I programatically extract the ID/Group from the ZD2 file (FX for G1Four and friends)
https://github.com/mungewell/zoom-zt2/blob/master/zoomzt2.py#L111
https://github.com/mungewell/zoom-zt2/blob/master/decode_effect.py

The ZDL file (FX for B1on) does not contain similar meta data, and this what we're trying to understand. They also do not appear to include XML to describe the parameter range/values.

The splitting of 32bits to a ID/Group matches how Zoom refer to FX in their 'ALLZD3.txt' files.

shooking, 2021-08-05:
Excellent info. The reason I split the 32bits is simply that's how they do it in the f0 ... 64 03 FXslot ... low high low high f7 message.

Ultimately the detail is intended to be hidden. Just we seem to need it if one wants to change FX in a slot.

Mostly the ZD2 files have consistent IDs across versions.
But the ZDLs - no so sure.

shooking, 2021-08-16:
OK so I made a bit more time over the weekend.
I can see now the B1On simply send new patch for FX insert.
There doesnt seem to be a mode - the hack I found that changes the FXID, but not the GID, is not likely to work.

So my only easy way to infer FX/GID on these types of pedals is:
inject into firmware.
setup my capture loop
edit patch to load "next" FX (as the B1XFour does)
zero out the parameters
What's left is the FXID and GID - but not much use for setting from Sysex
So likely better to just work out the 7 bit variant and snip that into the patch data - see if any checksums need fixing.

This seems to be how most of the common editors are doing it too.
Given I got Particle reverbs now on the GCE-3, I think I will play more with that.
It basically responds to Mungewell's zoom-zt2 with a little tweaking.
A few file format to reverse (G5n, G3n, B3n - likely same as G3n) - the "*1XFour's" are all known.

Closing this as it seems to be a pointless exercise (fun, but without Sysex to set the GID it is of no use).

closed 2021-08-16 by shooking

shooking, 2021-09-22:
Just to say the Zoom type is encoded in a ZDL itself, and hence in the firmware. It is bytes 0x40.to 0x47 read from the 00 00 00  00 start. It took me a while understand what @Barsik-Barbosik was saying about block size. So if a ZDL was more than 4090 bytes I had to remove the 6 bytes for prev/curr/next block.

Still not found Sysex on B1ON or MS-70 CDR to set FX SLOT. Easy on a G5 and thanks to @mungewell easy on the newer pedals.