Barsik-Barbosik/Zoom-Firmware-Editor #53: just a question: does it work on Linux?

peterjosvai, 2022-06-29:
hi, 
I'm not dumb, I hope at least, 
sorry for this trivial question, but I would only like to grab an old windows instance on one of my machines if it really is necessary...

from this:

```
Getting Started

Zoom Firmware Editor application is a self-executable jar-file. If you have Java JRE installed, then double-click on "ZoomFirmwareEditor.jar" should launch it.

If you dont have Java installed on your computer, you can download and run "ZoomFirmwareEditor-setup.exe". That setup application will install the compiled Windows executable with embedded JVM (created using Excelsior JET).
```

It seems to me you don't need windows, only Java 8, 
if anybody could confirm this, I'd get to flashing my B1On more lightheartedly :) 

thanks,

Peter

diovudau, 2022-09-27:
You can modify the firmware with java on linux, just java -jar it.
However, this program works by modifying an official windows firmware installer, which you then have to use.
That runs in WINE.

peterjosvai, 2022-09-27:
thank you :)

one more thing: to update the firmware I'll need the windows updater anyway, right?